grblHAL/RP2040 #16: Biqu SKR Pico

Deneteus, 2022-01-02:
Is the Biqu SKR Pico currently supported? 
https://www.biqu.equipment/products/btt-skr-pico-v1-0

Volksolive, 2022-01-03:
I don't think it is supported right now, @terjeio correct me if I am wrong. But BigTree Tech share the schematics on their Github: https://github.com/bigtreetech/SKR-Pico/blob/master/Hardware/BTT%20SKR%20Pico%20V1.0-SCH.pdf so from there you can create a board mapping and some modifications in the driver to support the 4 axis. I believe the first version of the Pico driver supported only 3 axis?! (before the shift register version). 

The 2 versions look quite good, I like the look of the armored version even though I don't think the protection plate will help in anyway with dust (if the board is used for milling cnc f.x) and the price since reasonably fair.

andrewmarles, 2022-01-03:
Major problem I see is that the step/dir are not arranged as needed by the existing PIO code.  Step pins need to be consecutive from what I understand.  I think this was also a problem with some arduino RP2040 modules as well.

Non-shift register PIO code supports up to six axes.  I have run 5 axes on the PicoBOB with no issues, very low jitter and high step rates.  It's actually the shift register version that is limited to 4 axes, I'd really like to dig in there and expand it to more but time is so finite these days.  However, as above the Biqu board splits up the step/dir so you would be limited to pure software stepping anyhow.  I haven't tried this on RP2040 (why would you when you have PIO?) so I am not sure if you can just map the pins and have it run but there is a non-zero chance it will work without too much fuss.

terjeio, 2022-01-03:
From a quick glance on the schematic I see that the PIO code cannot be used - and I cannot spot where the UART signals for the stepper drivers come from (they are TMC2209s?). So there will be quite a bit of work to get the driver modified for this board, doable but not easy without access to one.

Deneteus, 2022-01-03:
> 

Mine came in today and it was shipped before Christmas. 
[image]

Volksolive, 2022-01-04:
Oh yeah, I forgot that the stp/dir needed to be consecutive ... However, there might still be a way of using the PIO! each axis could use one PIO State machine (4 by per PIO and 2 PIO so 8 SM in total). The SM can be synched by a wait IRQ instruction, so all the main program has to do is to send the stp/dir data to each SM buffer and then set an IRQ to start all PIO SM at the same time. It might be possible to write the program using the OUT, SET or SIDE-SET functions so that 2 non consecutive pins can be used. That would not have been necessary if the Z axis stp/dir pin were consecutive like the other axis.

Deneteus, 2022-01-04:
I contacted Biqu to see about getting you some samples. I sent them a link to this conversation for reference.

terjeio, 2022-01-19:
FYI I have successfully added support for separate state machines for step generation, one per axis. Each uses 7 instructions so up to four axes by pio0 is possible, and an additional up to 3 by pio1. Currently I have only added support for 3 axes in the dev version I have... Stepper direction and enable signals are handled by regular GPIO.

Deneteus, 2022-01-20:
> FYI I have successfully added support for separate state machines for step generation, one per axis. Each uses 7 instructions so up to four axes by pio0 is possible, and an additional up to 3 by [...]

Thanks for the update. I'll test it this weekend and let you know what happens.

terjeio, 2022-01-20:
@Deneteus The change is not yet committed to github, and I do not have a map file for the board. Can you create one and upload here?

Volksolive, 2022-01-20:
@terjeio not sure how you did your PIO program but I just wanted to highlight the fact that all PIO State machines can run the same program. Because you said, each uses 7 instructions, but if it is all the same program for each state machines then you don't need to create it multiple times. Maybe that is already the way you've done it? 
Anyway since there is only 7 lines per SM and maximum program is 32 lines it isn't an issue here. But that might be helpful to know if space becomes limited.

terjeio, 2022-01-20:
> not sure how you did your PIO program

I have used the [current one](https://github.com/grblHAL/RP2040/blob/620124988015d7e72f8d8b347c54ae9d011caab1/driverPIO.pio#L70-L83) with one instance per axis and setting number of consecutive pins to 1.

> but I just wanted to highlight the fact that all PIO State machines can run the same program.

Perhaps my use of words is wrong? A PIO program is a state machine? Or is PIO peripheral == state machine? I regard a PIO program as a state machine.

> Because you said, each uses 7 instructions, but if it is all the same program for each state machines then you don't need to create it multiple times. Maybe that is already the way you've done it?

I have created multiple instances since each instance is bound to the pin(s) it uses. AFAIK a single instance cannot serve pins that are not consecutive:
```
#ifdef STEP_SINGLE_AXIS_SM
    uint32_t step_sm = 0;
  #ifdef X_STEP_PIO_PIN
    x_step_sm = step_sm++;
    pio_offset = pio_add_program(pio0, &step_pulse_program);
    step_pulse_program_init(pio0, x_step_sm, pio_offset, X_STEP_PIO_PIN, 1);
  #endif
  #ifdef Y_STEP_PIO_PIN
    y_step_sm = step_sm++;
    pio_offset = pio_add_program(pio0, &step_pulse_program);
    step_pulse_program_init(pio0, y_step_sm, pio_offset, Y_STEP_PIO_PIN, 1);
  #endif
  #ifdef Z_STEP_PIO_PIN
    z_step_sm = step_sm++;
    pio_offset = pio_add_program(pio0, &step_pulse_program);
    step_pulse_program_init(pio0, z_step_sm, pio_offset, Z_STEP_PIO_PIN, 1);
  #endif
#elif ...
```

Volksolive, 2022-01-20:
AFAIK, the PIO SM have each their own PC program counter, Out pin, Set pin and Side_set pin conf. So the program only tells the SM that a Out/Set or side_set is required, and the SM translate that into the right GPIO. So I believe if you only use pio_add_program once and reuse the same offset for each SM init it should also work.

terjeio, 2022-01-23:
Tentative map file and support code committed for this board, I have not tested TMC2209 communication but with some luck it may work. Pin mappings for control inputs and coolant has to be added - nut sure which pins are best suited.

@Volksolive Can you test adding the step pulse PIO program only once for this driver, I am afraid I do not understand how it can be done.

Volksolive, 2022-01-25:
@terjeio I have created a Pull request for the one PIO program only.

I think this schematic will explain how that is possible.

As you can see each SM has is own read access to the instruction memory, so they can all point to the same program and to different lines of this program asynchronously because each SM has its own program counter.

Also the program only refers to f.ex. in that case `out pins, 6` which does not point to any specific GPIO. This is because it is the SM that has its own pin configuration. So the same program on 2 diff. SM with 2 diff. pin configuration will give a different output.

[image]

The way the program is done now, the first SM generate the step before the next one and so on, so there is a slight delay between each axis steps. I guess this delay is negligible ?! But there could be a way to sync them. What do you think?

Also, reading at the code, I think there is a dir delay for every steps even when there is no dir changes, is that correct? And if so should we change that?

terjeio, 2022-01-26:
Thanks, I will test later.

> The way the program is done now, the first SM generate the step before the next one and so on, so there is a slight delay between each axis steps. I guess this delay is negligible ?! But there could be a way to sync them. What do you think?

If it takes 12 instructions to start a pulse then that is a ~100ns delay between motors, at 1000mm/s it will result in a 100nm skew. IMO nothing to worry about, and it will be far less than jitter from other bits of code - e.g. when a new step segment is pulled for execution.

> Also, reading at the code, I think there is a dir delay for every steps even when there is no dir changes, is that correct?

Yes.

> And if so should we change that?

No, if you worry about the skew fom starting the motor pulses separately then this should worry you more as the delay can be configured to be several microseconds.

terjeio, 2022-01-27:
I have refactored and cleaned up the code so please update before testing this board.
It should be possible to use the E0 with this version, either as a ganged axis or as A-axis.

Sebastian-MrBeam, 2022-03-02:
Hello, many thanks for the great work so far! I try to get the TMC-UART working on my BTT PICO SKR and I think I found a copy paste  bug - stopped me doing this.

serial.c line 448

```
static bool serial2Disable (bool disable)
{
   if(disable)
        hw_clear_bits(&UART2->imsc, UART_UARTIMSC_RXIM_BITS|UART_UARTIMSC_RTIM_BITS);       
    else
        hw_set_bits(&UART->imsc, UART_UARTIMSC_RXIM_BITS|UART_UARTIMSC_RTIM_BITS);    
}
```

**UART2** IRQ get disabled but **UART** IRQ get enabled    -> no UART 2 RX Interrupt no Data from TMC`s ;-)

terjeio, 2022-03-03:
@Sebastian-MrBeam - were you able to fix it by changing the code to enable UART2 RX irq?

Sebastian-MrBeam, 2022-03-03:
This was the first Bumper. No RX Data at all. With active UART2 RX IRQ data is received as expected. 
The second Bumper is the Pico HW FIFO Buffer under the SW Ringbuffer. To eliminate TMC OneWire  Reflex input  -  serial2Disable is called (.disable_rx = serial2Disable,) every time data is send to TMC´s. Unfortunately this only kill the IRQ handler filling the Ring-buffer, but all RX Data is still stored into HW FIFO. By reactivating RX IRQs to receive TMC data the HW FIFO is directly piped into the Ring-buffer -> no suppressing effect.
There are no helper Functions in Serial  or Stream  for control HW FIFO   (e.g.   .enable_FIFO  .clear_FIFO), for now I fixed it this way.
```
static bool serial2Disable (bool disable)
{
   if(disable){
        hw_clear_bits(&UART2->imsc, UART_UARTIMSC_RXIM_BITS|UART_UARTIMSC_RTIM_BITS);       
        uart_set_fifo_enabled(UART2_PORT, false);
       }
    else{
        hw_set_bits(&UART2->imsc, UART_UARTIMSC_RXIM_BITS|UART_UARTIMSC_RTIM_BITS);
        uart_set_fifo_enabled(UART2_PORT, true);
    }    
}
```
disabling the FIFO did not clean the first unhandled Byte  (only the FIFO Buffer itself). This Byte has to be cleared by reading it from UART.  I added this into Serial2RxFlush for now  - mixing HW and SW Buffer operations is no clean way at all - but works for now.

```
static void serial2RxFlush (void)
{
    while (uart_is_readable(UART2_PORT)) {
        uint32_t data = UART2->dr ;}
    rx2buf.tail = rx2buf.head;
    rx2buf.overflow = false;
}
```

I´m now able to do initial setup of the TMC´s, but get a hangup on M122 (TMC Config Report).
I spend some hours with Oszi and Logic Analyzer to find out what happens between Pico and TMC. Now with an established communication and some more hours it should be possible to get it completely working.

Sebastian-MrBeam, 2022-03-10:
@terjeio  I got the BTT-Pico board moving my gantry.  One question, is the TMC2209 part of the  Trinamic driver ever been tested with hardware? Is still to know if I don´t understand the code in some parts or it is compileable but never debugged/used or working fine for a different TMC Driver. Just to have a feeling how big changes should be necessary.

terjeio, 2022-03-10:
> One question, is the TMC2209 part of the Trinamic driver ever been tested with hardware?

I have tested with motors but not in a machine, this with STM32F4xx and LPC176x. AFAIK TMC2209 drivers are in use by others with BTT SKR boards.

> Just to have a feeling how big changes should be necessary.

For communication with the drivers? I guess that would be to debug/verify the RP2040 specific UART code. Enable only the X-axis and get that working first. Use `M122I` to reinitialize the driver when debugging.

Sebastian-MrBeam, 2022-03-11:
Thanks, this give me a feeling.  
With the changes above in     serial.c line ~448 static bool serial2Disable (bool disable)
and fixing the RP2040 HW FIFO behavior in    [tmc_uart.zip](https://github.com/grblHAL/RP2040/files/8230511/tmc_uart.zip)

Initial setup the TMC´s works fine for me (looks also good on Logic Analyser and Registerwise with Picoprobe debugging) 

Maybe someone could counter test  this on a BTT Pico SKR

The M122  output was a more strange Problem 
in Trinamic.c line ~1325   for reporting the current Settings, this Statement is used

`sprintf(append(sbuf), "%8d", stepper[motor]->get_config(motor)->current);
`

I don´t no why, but this statement kills the RP2040 with a   "decl_isr_bkpt isr_hardfault" - not handled IRQ what ever this mean???
The statement itself is valid -> debugger shows 500(mA) before hanging up in next step. 

To fix this behavior a copy of the trinamic_config_t  helps me out.

Trinamic.c line ~1325
```
sprintf(sbuf, "%-15s", "Set current");
        for(motor = 0; motor < n_motors; motor++) {
            if(bit_istrue(axes, bit(motor_map[motor].axis))){
                trinamic_config_t tmp_con; //->current;
                memcpy(&tmp_con, stepper[motor]->get_config(motor),sizeof(trinamic_config_t));
                sprintf(append(sbuf), "%8u", tmp_con.current);
                }
        }
        write_line(sbuf); 
```
This behavior also comes up on some other TMC Control Functions (e.g. change to stallguard mode) i work on now ->

Initial Motor setup         is working!
M122                              is working!
M122I                             is working!
M122 X S1                      in progress...
.
.
sensorless homing          ->Goal!

terjeio, 2022-03-14:
Try with moving `#pragma pack(push, 1)` at line 78 in front of the `trinamic_config_t` struct (to line 67):

https://github.com/terjeio/Trinamic-library/blob/1fe15f73a1235326a561305407bce30d53906034/common.h#L68-L78

Sebastian-MrBeam, 2022-03-14:
Great! Works! so only changes on RP2040 Driver files are needed (serial.c and tmc_uart.c)  - are you interested in a pull request?

terjeio, 2022-03-14:
> are you interested in a pull request?

Not needed as I have made the changes myself, will commit a bit later. BTW what did you change in _tmc_uart.c_?
I have made some changes to the map file as well as _driver.c_ - bugs in both.

Sebastian-MrBeam, 2022-03-15:
> I have made some changes to the map file as well as driver.c - bugs in both.

Is any of this Bugfixes related to Stepper.enable -  I used a workaround for my testing (Make P2 P7 P12 Aux Pins, they are init low and Steppers always enable).  If not I could do some testing there?

terjeio, 2022-03-15:
> Is any of this Bugfixes related to Stepper.enable

Yes. I'll commit the changes later today.

terjeio, 2022-03-15:
Changes just comitted. I have modifed the map file as well, it may still need tuning.
I did not add most of your changes to _tmc_uart.c_ as there was a bug in the tx buffer count function in _serial.c_ that required your extra delays as a workaround.

Sebastian-MrBeam, 2022-03-15:
No problem, integrate the HW-FiFo and Busy state into counting is such a smart and solid way to fix it - Far out of my mind - but nice to read.

closed 2022-09-27 by terjeio