grblHAL/RP2040 #19: Homing on RP2040 Pico board

jafgraphics, 2022-01-15:
Hi, I hope you can cast some light onto this.
I'm having an issue homing on a Pico board. The homing sequence alarms out with alarm 8. 
All parts of the homing process seem to work but not all together in a complete homing cycle. It appears something is setting an alarm between the limit striking and been able to pull off the switch.
I think it is some sort of config issue, see driver map below.

Below are the sequence of events.
First sequence is with the limit switch not pressed when homing started

1. The axis will run onto the limit switch at the seek rate.
2. Debounce delay observed
3. Direction of axis is reversed.
4. Pulses the axis for about 20ms and the alarms out with Alarm 8. 

Second sequence is with limit pressed when homing started

1. The axis moves off the limit switch at seek rate the distance set in Pull Off distance
2. Axis runs onto limit switch at feed rate.
3. Debounce delay observed
4. Direction of axis is reversed.
5. Pulses the axis for about 20ms and the alarms out with Alarm 8. 

Many thanks,
Simon

/*
  my_machine_map.h - driver code for RP2040 ARM processors
*/

#if TRINAMIC_ENABLE
#error Trinamic plugin not supported!
#endif

#if N_ABC_MOTORS > 0
#error "Axis configuration is not supported!"
#endif

// Define step pulse output pins.
#define STEP_PINS_BASE 6 // N_AXIS number of consecutive pins are used by PIO

// Define step direction output pins.
#define DIRECTION_PORT          GPIO_OUTPUT
#define X_DIRECTION_PIN         10
#define Y_DIRECTION_PIN         11
#define Z_DIRECTION_PIN         12
#define DIRECTION_OUTMODE       GPIO_MAP

// Define stepper driver enable/disable output pin.
#define STEPPERS_ENABLE_PORT    GPIO_OUTPUT
#define STEPPERS_ENABLE_PIN     14
#define STEPPERS_ENABLE_MASK    STEPPERS_ENABLE_BIT

// Define homing/hard limit switch input pins.  Currently configured so that X and Z limit pins are shared.
#define LIMIT_PORT                  GPIO_IN
#define X_LIMIT_PIN                 2
#define Y_LIMIT_PIN                 3
#define Z_LIMIT_PIN                 4
#define X_LIMIT_BIT                 (1<<X_LIMIT_PIN)
#define Y_LIMIT_BIT                 (1<<Y_LIMIT_PIN)
#define Z_LIMIT_BIT                 (1<<Z_LIMIT_PIN)
#define LIMIT_MASK                  (X_LIMIT_BIT|Y_LIMIT_BIT|Z_LIMIT_BIT)
#define LIMIT_INMODE 				        GPIO_MAP

// Define ganged axis or A axis step pulse and step direction output pins.
//#if N_ABC_MOTORS > 0
//#define M3_AVAILABLE
//#define M3_STEP_PIN             3
//#define M3_DIRECTION_PIN        (Z_DIRECTION_PIN + 1)
//#define M3_LIMIT_PIN            (Z_LIMIT_PIN + 1)
//#endif

// Define spindle enable and spindle direction output pins.
//#define SPINDLE_ENABLE_PORT     GPIO_OUTPUT
//#define SPINDLE_ENABLE_PIN      16
//#define SPINDLE_DIRECTION_PORT  GPIO_OUTPUT
//#define SPINDLE_DIRECTION_PIN   17

// Define spindle PWM output pin.
#define SPINDLE_PWM_PORT        GPIO_OUTPUT
#define SPINDLE_PWM_PIN         22
#define SPINDLE_PWM_BIT         (1<<SPINDLE_PWM_PIN)

// Define flood and mist coolant enable output pins.
//#define COOLANT_FLOOD_PORT      GPIO_OUTPUT
//#define COOLANT_FLOOD_PIN       26
//#define COOLANT_MIST_PORT       GPIO_OUTPUT
//#define COOLANT_MIST_PIN        27
//#define COOLANT_OUTMODE         GPIO_MAP

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define RESET_PIN               18
#define FEED_HOLD_PIN           19
#define CYCLE_START_PIN         20
#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         21
#endif
#define CONTROL_INMODE          GPIO_MAP

// Define probe switch input pin.
#define PROBE_PIN               0

//#if KEYPAD_ENABLE
//#define KEYPAD_STROBE_PIN       1
//#endif

terjeio, 2022-01-15:
Thanks, fixed now - plus another bug affecting hard limits when settings were changed.

jafgraphics, 2022-01-15:
Not sure the change you committed will actually fix the issue.
As I understand the code the only way to disable the hard limit interrupt during homing is if "homing" is True
when the function is entered "on" will either be "false" or "settings.limits.flags.hard_enabled" so if "settings.limits.flags.hard_enabled"=true and "homing"="true" the hard limit interrupt will still be enabled because the homing state is never considered.

 on = on && settings.limits.flags.hard_enabled; needs to be on=on && !homing;

terjeio, 2022-01-16:
The `homing` parameter is for switching to a second set of inputs if limit and home switches are separate or to output from the stepper drivers (e.g. when using Trinamic sensorless homing). The `on` parameter is for interrupt enable/disable.
When homing the `on` parameter is set to false [here](https://github.com/grblHAL/core/blob/9a904127b441ca3d08c38155190f1f235b8ef4dc/motion_control.c#L787) and reset to the configured hard limits state when homing is complete.

jafgraphics, 2022-01-16:
OK, makes sense now, wasn't obvious from reading the source but have now seen it in the HAL Structure docs. Thanks.

closed 2022-01-20 by terjeio