grblHAL/RP2040 #24: PIO step/dir shift register improvement + support of 8 motors

Volksolive, 2022-02-02:
@terjeio, as discussed in the other discussion, here is the .zip with 6 modified files:

- driver.c
- driver.h
- driverPIO.pio
- my_machine.h
- pico_cnc_map.h
- test_8axis_map.h

I have created a test mapping (test_8axis_map.h)
The grbl->config.h needs to be changed to N_AXIS 8 (or needs to be redefined in the mapping ?)
The pin mapping follows the pico_cnc board which I have updated 'map.h' to work with the new PIO code.

[PIO 8 Axis.zip](https://github.com/grblHAL/RP2040/files/7990108/PIO.8.Axis.zip)

I have tested with my DSO the signals for 4 motors config and 8 motors config, the signals looks good.
If you are willing to test on your shift register test rig that would be nice otherwise is there anybody with a Pico_CNC willing to test?

andrewmarles, 2022-02-02:
I have a pico board that can optionally take a 2nd shift register, I will see if I can find some time to test this.

Volksolive, 2022-03-01:
@terjeio I have finally received my Pico CNC board and will start testing my code on it. I have one question though, where should I declare that the board has 4 axis? Should that be in grbl/config.h? If that's the case, wouldn't it be better to declare it in my_machine.h instead?  or in the board mapping?

terjeio, 2022-03-01:
> Should that be in grbl/config.h?

Yes. Or in [CMakeLists.txt](https://github.com/grblHAL/RP2040/blob/master/CMakeLists.txt) - many symbols in the core are guarded by #ifndef ... #endif to allow them beeing set from the compiler command line.

> If that's the case, wouldn't it be better to declare it in my_machine.h instead? or in the board mapping?

IMO no and no. I do not want the core to reference symbols defined in the driver code.

closed 2025-12-25 by terjeio