grblHAL/RP2040 #25: Automatic tool probing issue

jafgraphics, 2022-02-03:
Hi,
I've set up a tool probe at G59.3 and TLO probing works OK but when I try to use automatic probing during a program the controller stops processing commands.
This is the process the controller goes through during a tool change.

1. Raise Z axis Up and waits for cycle_start press. ( Tool changed here)
2. Moves to the G59,3 location.
3. Performs probing cycle.
4. Moves Z axis up and moves back to the XY location where the tool change was called.
5. Pauses for 3-4 seconds ( Not sure why?)
6. Moves Z down to return cutter to original position.

Now the controller state is "IDLE" and nothing happens, I would assume the controller should continue running the program.
Pressing "cycle_start" does nothing.
Pressing "unlock" allows the program to continue but the toolchange block if marked as "go"

I've debugged the tool change sequence and everything seems OK, the only driver related issue I could find was that there was no hal.irq.enable or disable functions set. As I don't fully understand how the interrupts are handled in grblhal I couldn't possibly comment.

Any ideas?

andrewmarles, 2022-02-03:
This is likely this issue:
https://github.com/terjeio/ioSender/issues/195

For me was fixed in the latest commit to the core.

jafgraphics, 2022-02-03:
Sounds very similar, i'll update tomorrow and test.  Thanks.

jafgraphics, 2022-02-04:
Thanks, that fixed the issue.

closed 2022-02-04 by jafgraphics