grblHAL/RP2040 #34: "PI Pico W" access point problem

oke24, 2022-10-08:
@terjeio, as you mentioned in your reply 
https://github.com/terjeio/ioSender/issues/216#issuecomment-1257055165
**PI Pico W** is now supported by grblhal and your RP2040 driver - great !

But actually I'm not able to activate WiFi access on my new PI Pico W board (board is OK, e.g. demo "blink" is running with flashing LED).

In **CMakeLists.txt** I set
`option(ADD_WIFI "Add WiFi networking" ON)`

and in **my_machine.h**
```
#define WIFI_SOFTAP          1
#if WIFI_SOFTAP > 0
#define NETWORK_AP_SSID      "grblHAL_AP"
#define NETWORK_AP_PASSWORD  "grblhal"
#define NETWORK_AP_HOSTNAME  "grblHAL_AP"
#define NETWORK_AP_IPMODE    0 // 0 = static, 1 = DHCP, 2 = AutoIP
#define NETWORK_AP_IP        "192.168.5.1"
#define NETWORK_AP_GATEWAY   "192.168.5.1"
#define NETWORK_AP_MASK      "255.255.255.0"
#endif
```

Compliation was successsful 
```
[cmake] PICO platform is rp2040.
[cmake] Build type is Release
[cmake] PICO target board is pico_w.

[build] [100%] Built target grblHAL
[build] Build finished with exit code 0
```
but I get some warnings
```
[build] /home/pi/pico/RP2040/networking/networking.c: In function 'networking_get_services_list':
[build] /home/pi/pico/RP2040/networking/networking.c:91:13: warning: 'strncat' specified bound depends on the length of the source argument [-Wstringop-overflow=]
[build]              strncat(list, service_names[idx], strlen(service_names[idx]) - 1);
[build]              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] /home/pi/pico/RP2040/networking/networking.c:91:47: note: length computed here
[build]              strncat(list, service_names[idx], strlen(service_names[idx]) - 1);
[build]  
                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

and

```
[build] /home/pi/pico/RP2040/webui/args.c: In function 'webui_get_bool_arg':
[build] /home/pi/pico/RP2040/webui/args.c:56:12: warning: 'strncpy' specified bound depends on the length of the source argument [-Wstringop-overflow=]
[build]          if(webui_get_arg(argc, argv, strncpy(tmp, arg, strlen(arg) - 1)))
[build]             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] /home/pi/pico/RP2040/webui/args.c:56:56: note: length computed here
[build]          if(webui_get_arg(argc, argv, strncpy(tmp, arg, strlen(arg) - 1)))
[build]                                                         ^~~~~~~~~~~
```

and very, very lot of
```
[build] <command-line>: warning: "NDEBUG" redefined
[build] <command-line>: note: this is the location of the previous definition
```

After copying the grblhal.uf2 file to PI Pico W there is no WiFi access point 
`grblHAL_AP`

So I'm not able to connect via WiFi. 
Do you have some hints how to make it run ? 

An additional question: what is this for ? 
```
// The following symbols have the default values as shown, uncomment and change as needed.
//#define NETWORK_HOSTNAME     "grblHAL"
//#define NETWORK_IPMODE       0 // 0 = static, 1 = DHCP, 2 = AutoIP
//#define NETWORK_IP           "192.168.1.210"
//#define NETWORK_GATEWAY      "192.168.1.1"
//#define NETWORK_MASK         "255.255.255.0"

```
Is it possible to connect via WiFi to my home network, get an local IP from my home network and is connectable with iosender, or is this only useable for boards with ethernet cable connection ?

Regards, Olaf

terjeio, 2022-10-09:
I do not get the warnings you see, my SDK verison is 1.4.0.

AP mode is not working properly, marked WIP (Work In Progress) in _my_machine.h_. I can connect but get an error after a short while on my Win7 laptop. Station mode is working for me. It could be that your settings has not been initialized properly, changed default values are not always propagated to settings (and should not be) - either set them manually or issue `$RST=*` to reset all to default values.

> Is it possible to connect via WiFi to my home network, get an local IP from my home network and is connectable with iosender

Yes. And the WebUI seems to be working as it should too (not tested in a real machine).
I have one issue remaining related to SSDP discovery - searh mode is not working. This is due to issues with the SDK which hopefully will be fixed in a newer version. There is a pending pull request for it.

Finally, check your settings when connected via USB. $70, 73, 74 and 75 has to be correct. `$I` (or Help > About in ioSender) will show the IP address when connected.

terjeio, 2022-10-09:
AP mode is working provided that settings are correct...
The AP mode IP address is hardcoded in the SDK and has to be set to 192.168.4.1 (setting $312) for it to work. I do not yet know if it is possible to have it as a run-time setting so perhaps I have to remove $312. There is a bug in $314 handling as well, it reports the station mask. Setting $314=255.255.255.0 and restarting makes AP mode work provided that $76 (AP SSID) and $77 (AP password) is set correctly.

FYI the WebUI is not yet working in AP mode, I'll commit a fix for that and the $314 bug later.

oke24, 2022-10-09:
Cause of the warnings are compile time options.

With setting NDEBUG from 0 to 1 in CMakeLists.txt
`target_compile_definitions(grblHAL PUBLIC NDEBUG=1)`
I get rid of the NDEBUG warnings.

The warnings in networking/sha1.c.obj and webui/args.c.obj
```
[build] [ 21%] Building C object CMakeFiles/grblHAL.dir/networking/sha1.c.obj
[build] /home/pi/pico/RP2040/networking/networking.c: In function 'networking_get_services_list':
[build] /home/pi/pico/RP2040/networking/networking.c:91:13: warning: 'strncat' specified bound depends on the length of the source argument [-Wstringop-overflow=]
[build]              strncat(list, service_names[idx], strlen(service_names[idx]) - 1);
[build]              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] /home/pi/pico/RP2040/networking/networking.c:91:47: note: length computed here
[build]              strncat(list, service_names[idx], strlen(service_names[idx]) - 1);
[build]                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~
```
and
```
[build] [ 25%] Building C object CMakeFiles/grblHAL.dir/webui/args.c.obj
[build] /home/pi/pico/RP2040/webui/args.c: In function 'webui_get_bool_arg':
[build] /home/pi/pico/RP2040/webui/args.c:56:12: warning: 'strncpy' specified bound depends on the length of the source argument [-Wstringop-overflow=]
[build]          if(webui_get_arg(argc, argv, strncpy(tmp, arg, strlen(arg) - 1)))
[build]             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build] /home/pi/pico/RP2040/webui/args.c:56:56: note: length computed here
[build]          if(webui_get_arg(argc, argv, strncpy(tmp, arg, strlen(arg) - 1)))
[build]                                                         ^~~~~~~~~~~
```

are generated when switching ADD_WIFI form OFF to ON in CMakeLists.txt
`option(ADD_WIFI "Add WiFi networking" ON)`

SDK version is 1.4.0, grblhal version and Networking services plugin from today. Perhaps the warning results from a compile time option not set, warning message says [-Wstringop-overflow=] ?

When switching to AP mode with  
`option(ADD_WIFI "Add WiFi networking" ON)`
and
`#define WIFI_SOFTAP          1`
I can't see the AP "grblHAL_AP" in my WiFi network list.  So this doesn't work for me.

I have to note that I'm testing only with the new PI Pico W board. 
PicoBOB and Mach 3 CNC controller are not connected (because they are part of the running live CNC environment which I don't will brake for these first tests with the new PIO Pico W board). 

The question is if I can test "WiFi station-" and "WiFi AP- mode" standalone with only PI Pico W connected to my home network and ioSender on a Windows machine. (when connecting ioSender via USB to PI Pico W I get only a permanent E-Stop error) .

terjeio, 2022-10-09:
> when connecting ioSender via USB to PI Pico W I get only a permanent E-Stop error

Invert the [E-stop control signal](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) in ioSender or connect the pin to ground to fix this.

> I can't see the AP "grblHAL_AP" in my WiFi network list. So this doesn't work for me.

Did you change the settings or reset them to default as described above? If not then it will not work. The settings are in the _Networking_ and _WiFi_ groups.

I have just committed an update with fix for the $314 bug and fixes to make AP mode work with the WebUI. Warnings can be ignored for now? - I do not get them so I have to check a bit more to find out why you get them.

oke24, 2022-10-09:
With the new fix I get some more warnings because of the standard NETWORK_AP_GATEWAY - IP in grblHAL.
In grbl/driver_opts.h the IP is hard coded to 192.168.**5**.1, in RP2040/my_machine.h set to 192.168.**4**.1
This results in warnings like
`warning: "NETWORK_AP_GATEWAY" redefined`

```
[build] In file included from /home/pi/pico/RP2040/driver.h:38,
[build]                  from /home/pi/pico/RP2040/usb_serial.c:41:
[build] /home/pi/pico/RP2040/grbl/driver_opts.h:381: warning: "NETWORK_AP_GATEWAY" redefined
[build]  #define NETWORK_AP_GATEWAY      "192.168.5.1"
[build]  
[build] In file included from /home/pi/pico/RP2040/driver.h:35,
[build]                  from /home/pi/pico/RP2040/usb_serial.c:41:
[build] /home/pi/pico/RP2040/my_machine.h:101: note: this is the location of the previous definition
[build]  #define NETWORK_AP_GATEWAY   "192.168.4.1"  // Do not change!
```
How can I fix this ?

terjeio, 2022-10-09:
> How can I fix this ?

Leave it commented out in [my_machine.h](https://github.com/grblHAL/RP2040/blob/68a34a23461ae69269b8e19598e71dee3ebde4ff/my_machine.h#L101), I have hardcoded the address in [wifi.c](https://github.com/grblHAL/RP2040/blob/68a34a23461ae69269b8e19598e71dee3ebde4ff/wifi.c#L949-L953).

oke24, 2022-10-09:
Thank you very much ! First tests looks good - I can connect from ioSender both in WiFi station mode with an IP from my home network and in WiFi AP mode with the hardcoded 192.168.4.1.  

Next I will test WebUI with WiFi station and WiFi AP mode.

oke24, 2022-10-09:
Unfortunately I get another compile error when activating WebUI.

WebUI plugin README.md in the grblHAL docu says
`Enable WebUI support by uncommenting #define WEBUI_ENABLE 1 in my_machine.h and recompile/reflash.`

When I do this in my_machine.h I get several errors like

```
[build] [ 11%] Building C object CMakeFiles/grblHAL.dir/i2c.c.obj
[build] In file included from /home/pi/pico/RP2040/flash.c:37:
[build] /home/pi/pico/RP2040/driver.h:203:2: error: #error SD card plugin not supported!
[build]  #error SD card plugin not supported!
[build]   ^~~~~
[build] gmake[2]: *** [CMakeFiles/grblHAL.dir/build.make:147: CMakeFiles/grblHAL.dir/flash.c.obj] Error 1
[build] gmake[2]: *** Waiting for unfinished jobs....
[build] In file included from /home/pi/pico/RP2040/serial.h:27,
[build]                  from /home/pi/pico/RP2040/serial.c:28:
[build] /home/pi/pico/RP2040/driver.h:203:2: error: #error SD card plugin not supported!
[build]  #error SD card plugin not supported!
[build]   ^~~~~
[build] In file included from /home/pi/pico/RP2040/usb_serial.c:41:
[build] /home/pi/pico/RP2040/driver.h:203:2: error: #error SD card plugin not supported!
[build]  #error SD card plugin not supported!
[build]   ^~~~~
[build] gmake[2]: *** [CMakeFiles/grblHAL.dir/build.make:108: CMakeFiles/grblHAL.dir/serial.c.obj] Error 1
[build] gmake[2]: *** [CMakeFiles/grblHAL.dir/build.make:121: CMakeFiles/grblHAL.dir/usb_serial.c.obj] Error 1
```

RP2040/driver.h:203: 
```
#if SDCARD_ENABLE && !defined(SD_CS_PIN)
#error SD card plugin not supported!
#endif
```

but SD card plugin is not defined in my_machine.h

```
//#define WIFI_ENABLE          0 // Do NOT enable here, enable in CMakeLists.txt!
#define WIFI_SOFTAP          1 // Use Soft AP mode for WiFi. NOTE: WIP - not yet complete!
#define WEBUI_ENABLE         1 // Enable ESP3D-WEBUI plugin along with networking and SD card plugins. Requires WiFi enabled.
//#define WEBUI_AUTH_ENABLE    1 // Enable ESP3D-WEBUI authentication.
//#define WEBUI_INFLASH        1 // Store WebUI files in flash instead of on SD card.
//#define SDCARD_ENABLE        1 // Run gcode programs from SD card.
```
I do a grep over the whole sources in RP2040 directory but I can't find a place in source code where SDCARD_ENABLE is set to 1.  Hmm ....

terjeio, 2022-10-10:
> I do a grep over the whole sources in RP2040 directory but I can't find a place in source code where SDCARD_ENABLE is set to 1. 

#define WEBUI_ENABLE         1 // Enable ESP3D-WEBUI plugin **along with networking and SD card plugins**. Requires WiFi enabled.

Enabled [here](https://github.com/grblHAL/core/blob/2f42b7226bc9c70b98d612ae1c46576b718153f5/driver_opts.h#L295-L299).

Enabling WebUI without a place to store gcode is not very useful?

terjeio, 2022-10-10:
The Pico (and now the Pico W) _driver_ supports SD card and other options, but not neccesarily all _board maps_ do as pins may be allocated differently - it is a trade-off what the board designer want to support as the limited number of pins is a hard constraint. I am testing with a prototype Pico CNC board that has [pins allocated for the SD card interface](https://github.com/grblHAL/RP2040/blob/68a34a23461ae69269b8e19598e71dee3ebde4ff/pico_cnc_map.h#L157-L165) and thus can be used with the WebUI. The WebUI can only execute gcode stored on a local SD card as the Pico processor does not have the amount of memory needed (except for small programs), thus the SD card dependency.

oke24, 2022-10-10:
Thanks for explanation ! Because I actually use the PicoBOB 
```
my_machine.h:
`#define BOARD_PICOBOB`
```
in combination with PI Pico I got the SD error.

Countertesting with 
`#define BOARD_PICO_CNC`
there are no errors during compilation.

terjeio, 2022-10-10:
Looking at the pin map it _might_ be possible to connect a SD card to the PicoBOB to pins 0, 1 ,2 and 7. You will loose motor enable for Y and Z and the UART input. Motor for all three motors enable can be connected to X enable. I do not know if this it electrically feasible with the card - or if it will work...