grblHAL/RP2040 #44: Aux Out config and errors

MadTooler, 2022-12-07:
I have setup my_machine_map.h as a copy of the generic plus some aux out pins (I also commented out the Flood and Mist pin defines since I wanted to use those pins for the Aux):

```
#define AUXOUTPUT0_PORT         GPIO_OUTPUT
#define AUXOUTPUT0_PIN          16
#define AUXOUTPUT1_PORT         GPIO_OUTPUT
#define AUXOUTPUT1_PIN          17 
```

When I check $pins, it shows the aux outputs, but I do not see the "P" increments at the end.
```
>>> $pins
[PIN:18,Emergency stop]
[PIN:19,Feed hold]
[PIN:20,Cycle start]
[PIN:22,Probe]
[PIN:9,X limit min]
[PIN:10,Y limit min]
[PIN:11,Z limit min]
[PIN:PIO2,X step]
[PIN:PIO3,Y step]
[PIN:PIO4,Z step]
[PIN:PIO5,N/A]
[PIN:PIO6,N/A]
[PIN:PIO7,N/A]
[PIN:5,X dir]
[PIN:6,Y dir]
[PIN:7,Z dir]
[PIN:8,Stepddpers enable]
[PIN:15,Spindle PWM]
[PIN:13,Spindle on]
[PIN:14,Spindle direction]
[PIN:16,Aux out 0]
[PIN:17,Aux out 1]
ok
```
When I run M64 P0, it throws an error
```>>> M64 P0
[Error] An error was detected while sending 'M64 P0': (error:20) Unsupported or invalid g-code command found in block. Streaming has been paused.
```
Am I missing something or is there an issue here?

Thanks.

terjeio, 2022-12-07:
Have you added `#define HAS_IOPORTS` to the map file?

MadTooler, 2022-12-07:
> 

That is what I was missing.  Sorry to be a pest.  Thanks!

I looked around a lot and kept missing that requirement.  Please add it to the my_machine.h template for others.

closed 2022-12-07 by MadTooler