grblHAL/RP2040 #45: GRBLHal freezes up when streaming many lines from pyserial

chris2560, 2022-12-16:
I'm writing a milling machine controller program, which communicates with GRBLHal on a PiCOBOB via USB.  (I'm running Fedora 34 and pyserial 3.5, the program is running under IDLE 3.9.13)

The machine controller I'm building accepts Fanuc or Haas like gcode and processes it on the fly to output GRBL or GRBLHal compatible GCode.

For some obscure reason, the communication randomly freezes up (I don't know if the machine motion continues or not - my machine isn't yet fully CNC).

I thought earlier it was the serial read buffer overflowing but the status reports always see to show at least 48 characters free space, each line I'm streaming when it freezes are 11 characters long formatted as:

> N100G00Z-1.1;

(The line number is so my program can work out from the status reports which line is being executed - the 'ok' could be in response to a gcode command or a status report command as far as I'm aware)

I'm certain the freeze occurs as data is written to serial because I made the program print before and after sending each line, and when it froze it had only printed start, not end also.  I don't know if pyserial is at fault or if GRBLHal is at fault.  This is my sending code:

> print("Start", end="");
      self.serialConnection.write(bytes(data+"\n", "utf8"));
      print("End", end="\n");
      
Any ideas would be appreciated
Thanks
Chris

terjeio, 2022-12-17:
If you connect with a terminal program do you get a status report when sending `?` after the freeze?

> self.serialConnection.write(bytes(data+"\n", "utf8"));

utf8 is not the correct charset to use, (7-bit) ASCII is - with some 8 bit characters supported.

chris2560, 2022-12-17:
Hi
Thanks for the reply,
I've just tried sending a ? after a freeze from a terminal 

> echo "?" > /dev/ttyACM4

and with another terminal open to read the output 

> cat /dev/ttyACM4

and both seemingly time out rather than sending or receiving any data successfully, but they don't return any errors either.

Under normal circumstances (when not frozen) the response I get to the same commands is a status report, and lots of error 9s

(The UTF8 encoding is not sending wierd characters, just sending gcode and control signals, I used the argument of utf8 as this function also handles sending control characters to GRBL)

chris2560, 2022-12-17:
Seem to have got to the bottom of the freezing up though just now:

The freezeup occurs as the status report ? is sent, and the buffer overflows.

The question this raises, is why do control fill space in the serial buffer when they are instantly processed? 
(this is not just a problem with GRBLHal, as the same occurs on GRBL on an Arduino Nano, except that I think the system ignores characters rather than freezing)

terjeio, 2022-12-18:
> The freezeup occurs as the status report ? is sent, and the buffer overflows.

Which buffer? Input or output?

> The question this raises, is why do control fill space in the serial buffer when they are instantly processed?

I do not really understand what you mean by this. Real-time commands are not buffered but any output generated by them is. And output that would cause a buffer overflow causes blocking while waiting for room.

> (this is not just a problem with GRBLHal, as the same occurs on GRBL on an Arduino Nano, except that I think the system ignores characters rather than freezing)

Input buffer overflow should be avoided by the sender side, either by waiting for the "ok" repsonse for each block sent or by using the character counting algorithm as overflowing characters are lost. Sending data via a native USB stack might be a bit different though as there will be additional buffers used by the stack. Note the Arduino Grbl has a USB/UART chip so no USB stack on the MCU - plain UART comms is used.

chris2560, 2022-12-18:
Thanks for the reply, sorry I didn't explain very clearly about the buffer overflow.

The buffer I'm talking about is the serial input buffer which the status reports tell me how much space is free in.

- If there is space in the planner buffer, the overflow never occurs.
- If the planner buffer is full, and a command or more are waiting in the serial read buffer, then all realtime ? commands also take up space in the serial read buffer, even though they are being processed instantly, they hog the space.

To demonstrate this issue:

1. send a straight line command followed by a curved path command which takes time to execute, this instantly fills the planner buffer

> G01X0.Y0.F100;
> G02X10.Y10.I10.J0.F200;

3. continue to send ? requests
4. the output looks something like this:

> Sending: ?...sent
> Recieved: <Run|MPos:9.772,9.772,0.000|Bf:0,983|FS:100,0>
> Sending: ?...sent
> Recieved: <Run|MPos:9.736,9.736,0.000|Bf:0,982|FS:100,0>
> Sending: ?...sent
> Recieved: <Run|MPos:9.699,9.699,0.000|Bf:0,981|FS:100,0>
> Sending: ?...sent
> Recieved: <Run|MPos:9.666,9.666,0.000|Bf:0,980|FS:100,0>

5. the input buffer space seems to now be slowly filling up, if that command happens to take a very long time to execute and my program asks for the position 10 times a second so it can keep the tk interface up to date (Fanuc machines seem to update the display about 10 times a second), the whole lot eventually freezes up.

This error shouldn't be ignored or assumed to be completely unlikely - it will only take a GCode program to nearly fill the planner buffer with a series of straight lines and then add a curved line command, and if it takes > 100.3 seconds to execute, then the system will freeze.

When both buffers are full, GRBL on an Atmega 328 seems to simply process the realtime status report request and output the status report, but then throw away the excess command which doesn't fit in the buffer, wheras GRBLHal simply freezes up if both buffers fill to the brim, this is probably the difference you mention caused by the pico including a USB stack.

I agree that my program must not send gcode if the buffers are full, but I don't understand why realtime requests seem to hog buffer space.  (I probably ought to ask the question to the developers of GRBL as it's GRBL which seems to cause the problem)

I've just been fiddling around with GRBL to see why everything ends up on the serial input buffer (in protocol.c, the main input handling loop) but I can't work out how to prevent realtime commands being buffered, somehow they get added to a char array called 'line', which I think then gets buffered??, yet if I wipe the line char array 'line[0] = 0' then it all still gets buffered.  (maybe its all buffered before this loop executes)

chris2560, 2022-12-19:
Hi,
The buffer I'm meaning is the GRBLHal serial input buffer.  The GRBL/GRBLHal serial output buffer is regularly polled (every 50 milliseconds) and emptied by my program.

I've spent some more time investigating the seemingly buffering of realtime commands:

- you are quite right, realtime commands are not buffered, (they are 1 character each and don't land on the buffer, they simply set flags which are then handled at each iteration of the protocol.c main loop).
- my program was sending the status report realtime command and a \n character each time
- the \n was getting buffered and hogging serial input buffer space, eventually causing the overflow freezeup.

(I only realised the real problem as I delved into the depths of GRBL and the serial input Interrupt Service Routine, and saw that the trailing \n would simply be added to the buffer.  I tried sending the ? raw and it worked properly, the test script I wrote then shows the buffer never getting any fuller even when the planner buffer is full).

As you mention about the difference in USB stack/UART chip, yes it appears I can send as much data as I wish to the Arduino nano, it simply ignores input if there's no space in the serial input buffer, wheras the Pico freezes up, (this is why I never encountered the freezing up even though I was watching the serial input buffer space shrinking on the Arduino Nano before I ever tried GRBLHal).

My apologies for wasting your time when in reality my program was at fault, and many thanks for taking the time to help.

closed 2023-04-29 by terjeio