grblHAL/RP2040 #46: Problems with the G10 command

Trev4G, 2023-01-06:
I am using bCNC to drive a Pi Pico board. My initial tests attached to my mill showed I can move the steppers as expected and even run an .nc file to move the axis around. However, I then trid to zero the WCS but this failed.Changing the axis to fixed values also failed. I traced this to the G10L20... command. When tried this with minicom it confirmed my suspicions. When a G10L20 command is issued I get back an 'ok' but the WCS value remains unchanged.
For example:

G10L20P1X0Y0Z0
ok
$#
[G54:21.000,-8.001,-5.000]
[G55:20.999,-8.000,-5.000]
[G56:0.000,0.000,0.000]
[G57:0.000,0.000,0.000]
[G58:0.000,0.000,0.000]
[G59:0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000,0.000,0.000]
[PRB:0.000,0.000,0.000:0]
ok

I checked I am using G54. Is there something I am missing?

terjeio, 2023-01-06:
What is your $10 setting value? 511 is the default.

Here are the flags available, many more than available in legacy Grbl:

```
$$=10
    0 - Position in machine coordinate (1)
    1 - Buffer state (2)
    2 - Line numbers (4)
    3 - Feed & speed (8)
    4 - Pin state (16)
    5 - Work coordinate offset (32)
    6 - Overrides (64)
    7 - Probe coordinates (128)
    8 - Buffer sync on WCO change (256)
    9 - Parser state (512)
    10 - Alarm substatus (1024)
    11 - Run substatus (2048)
Specifies optional data included in status reports.
```

Trev4G, 2023-01-06:
Thanks for your reply. Setting $10 to 511 has solved the problem. 

I have a CNC3018 router, with legacy grbl, that I am using to give me a guide while I modify my Mill. I copied its value of $10=3. All the comments I have seen online suggest either $10=1 or $10=3.

Is there a settings guide for grblHAL ?  I see settings that I have no description for.
I have already had problems with $14. The only online guides to settings I have found do not mention $14 with one exception which has the wrong description.

Many thanks for the solution. Now off to progress the Mill conversion....

terjeio, 2023-01-06:
> All the comments I have seen online suggest either $10=1 or $10=3.

grblHAL settings are not neccesarily compatible with legacy Grbl. E.g. many compile time options in legacy Grbl are now run time configurable. It is possible to compile grblHAL in [compatibility modes](https://github.com/grblHAL/core/wiki/Compatibility-level), but that will lead to loss of (some of the new) functionality.

> Is there a settings guide for grblHAL ?

See the [Wiki](https://github.com/grblHAL/core/wiki/Additional-or-extended-settings). If you are running on Windows you may try [ioSender](https://github.com/terjeio/ioSender) for configuring, it has an advanced settings UI built using information from the controller.

`$$=<setting number>` will provide some info about the setting and will always by up to date related to the active configuration.

closed 2023-01-06 by Trev4G