grblHAL/RP2040 #47: Laser mode - laser enable pin not triggered as expected at G0/G1

MadTooler, 2023-01-25:
Hello again.  I am having fun applying grblHAL via a Pico RP2040 to a 2 axis CNC running in laser mode.  Not a laser, but similar intended function where I need the "spindle" to stop upon G0 commands and start upon G1 or related commands.

I only need on/off to my spindle.  I do not need PWM.  I am using the spindle/laser enable pin as my only spindle output.  Perhaps that is related to the issue I am seeing.

The issue... From the beginning of the gcode, all G0 commands after M3 are performed with the spindle/laser pin enabled.  Is this as intended or is this an issue?  Should I be using the PWM spindle pin instead and if so, can that be set to non PWM output?

Thanks!

terjeio, 2023-01-26:
Try using M4 and set `$9=3`.

MadTooler, 2023-01-27:
Thanks for the reply and suggested solution.

> Try using M4 and set `$9=3`.

It looks like those settings may be working.  I need to test it a bit more.  

I could not find what the value of 3 in setting $9 is doing.  Can you explain or point me to the wiki that I possibly overlooked?

Also, I have been tweaking the default values in the firmware so future builds load up proper without going through all the $ fixes.  Where do I find the $9 value to adjust within the code?

Thanks again!  My prototype machine is making some impressive projects so far.  I hope to one day be able to share what it is doing and make it available for others as well.

MadTooler, 2023-01-27:
I saw the options under $$9 for 0 and 1.  I did not see anything there or in the code for what it does when I send it 3 as you suggested.  It seems to be working, but I do not understand why which always makes me worried it will at some point do something unexpected at usually the worst possible time.  To confirm, it is correct at $9=3?

terjeio, 2023-01-27:
> To confirm, it is correct at $9=3?

Yes, for your use-case I belive it is.

The default value in config.h is used after all. The search for it did not return all results.

closed 2023-03-03 by MadTooler