grblHAL/RP2040 #62: RP2040 Zero

Juank23, 2023-05-14:
Good morning, my friend I acquired a pico zero a few days ago and I generated the binary through the Web Builder page, and it works but the pins are different to the ones that my board has, I downloaded the code for vscode but I have not been able to compile it, thanks for your help.
[image]

Juank23, 2023-05-14:
thanks for your ptonta respusta, it is necessary to install something to the vscode so that it can compile the firmware for the pico zero....
[image]

terjeio, 2023-05-15:
> it is necessary to install something to the vscode so that it can compile the firmware for the pico zero....

Yes, a toolchain and the SDK has to be installed, follow the link in the [readme](https://github.com/grblHAL/RP2040/blob/master/README.md). And there are relevant links in the search I provided above.

andrewmarles, 2023-05-15:
Under windows I have found this script to work well:

https://github.com/raspberrypi/pico-setup-windows

Juank23, 2023-05-18:
good day, I tried all the steps but it still doesn't work, is there any way to change the pinout for my pico Zero I remain attentive thank you.....

terjeio, 2023-05-19:
Create a file name _my_machine_map.h_ with the pin map in the same directory as the other maps (I would start with a copy of [generic_map.h](https://github.com/grblHAL/RP2040/blob/master/generic_map.h)) and edit it to suit your requirements, then enable it by [uncommenting this line](https://github.com/grblHAL/RP2040/blob/078836e5ff6c2366b095f2602e07ca16b6d09d82/my_machine.h#L31) before compiling.

Juank23, 2023-05-20:
after a few days I managed to compile the code, thank you very much for your excellent work.

terjeio, 2023-05-21:
Good to hear that, have fun with your machine!

closed 2023-05-21 by terjeio