grblHAL/RP2040 #63: Fresh install on BOARD_PICO_CNC goes straight into alarm on boot

leofro, 2023-05-20:
Hello,

after uncommenting the line #define BOARD_PICO_CNC in my_machine.h and flashing a basic Raspberry Pi Pico with the latest grblHAL, I get "<Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PYZHSE|Ov:100,100,100>" after sending ? to the Pico.
$I prints
`[VER:1.1f.20230519:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,ES,TC,SED,RTC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:RP2040]
[DRIVER VERSION:230515]
[DRIVER OPTIONS:SDK_1.5.0]
[BOARD:PicoCNC]
[AUX IO:5,8,0,0]
ok`

$X prints
`error:50`

I tried the following:
$RST=$ prints
`[MSG:Restoring defaults]
ok
[MSG:Emergency stop - clear, then reset to continue]`

After that ? prints
`<Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PYZHSE|H:0>`

and $X:
`error:79`
and it stays at error 79 even after performing a reset via ctrl-x or powering the Pico off and on again.
However, ? changes output to <Alarm|MPos:0.000,0.000,0.000|Bf:35,1023|FS:0,0|Pn:PXYZHSE|Ov:100,100,100> after performing a reboot.

I'm kind of lost regarding the error codes and output of ?..

Nothing but the USB cable is connected to the Pico at this point. What could I have done wrong?
I thought alarms are disabled by default and I have to enable features like homing to activate alarm functions..
Any help is appreciated!

Thank you and kind regards
Leo

terjeio, 2023-05-20:
See the [core Wiki for first run info](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).

leofro, 2023-05-20:
@terjeio : Thank you so much!!
As stated in the mentioned wiki, grblHAL is awaiting low state on inputs EStop/Reset, Cycle Start and Feed Hold.
 `$14=70` (invert these three inputs) did the trick for me to get things running without any periphery.
Again, thank you very much :))

closed 2023-05-20 by leofro