grblHAL/RP2040 #64: Pico Zero

Juank23, 2023-05-21:
Good morning, is there any way to flash the RGB led that has the pi zero ,
that indicates when it is working, I remain attentive
[image]

terjeio, 2023-05-22:
You may add a plugin for that - [here](https://github.com/grblHAL/Templates/tree/master/my_plugin/stm32F411blinky) is how I did it for a STM32F411 board. The WS2812 is a bit harder to interface to as it has a serial interface...

Juank23, 2023-05-23:
Good morning, friend add those lines and edit the pin but at the time of compiling gives many errors, then I send you an image thank you
[image]

terjeio, 2023-05-23:
The code that does not compile is STM32F411 specific and for a single LED  switched on/off by a logic level. You have to replace that with code suitable for the RP2040 and the type of LED used. You may find suitable code by [searching for it](https://duckduckgo.com/?t=ffab&q=RP2040+WS2812&ia=web).

terjeio, 2023-05-23:
I found [this](https://www.waveshare.com/wiki/RP2040-Zero#Demo_Codes), it has a plain C example.

terjeio, 2023-05-23:
Oops, I forgot that you have to [enable my_plugin.c in the build](https://github.com/grblHAL/RP2040/blob/cbf8267b5e82c429008d08d64f2d0892b8b36be1/CMakeLists.txt#L12) for this MCU.
You can check if the plugin is active by sending `$I` - it will add itself to the output when it is.

A plain LED will work when switching a GPIO pin on/off but not a WS2812, it uses serial commands with [strict timing requirements](https://duckduckgo.com/?t=ffab&q=+WS2812+timing&ia=web)...

Juank23, 2023-05-23:
I already did what you told me but it doesn't work, can you help me with an example to make it work in your code?
[image]
[image]

Juank23, 2023-05-23:
friend gives errors, and just modify the line that is OFF and without modifying the plugin also gives error, and does not work, I have another normal pi 2040 that has the led on pin 25 and also does not work ......
[image]

terjeio, 2023-05-23:
Ok, I have fixed and tested the code for you - this works with a plain LED (with an appropriate series resistor in place):

```
/*

  my_plugin.c - user defined plugin that blinks the LED on a STM32F411 Blackpill

  Part of grblHAL

  Public domain

*/

#include "driver.h"
#include "hardware/gpio.h"

const uint LED_PIN = 25;
 
static on_report_options_ptr on_report_options;
static on_execute_realtime_ptr on_execute_realtime;

// Add info about our plugin to the $I report.
static void on_report_my_options (bool newopt)
{
    on_report_options(newopt);

    if(!newopt)
        hal.stream.write("[PLUGIN:Blink LED v1.00]" ASCII_EOL);
}

static void blink_led (sys_state_t state)
{
    static bool led_on = false;
    static uint32_t ms = 0;

    if(hal.get_elapsed_ticks() >= ms) {
        ms = hal.get_elapsed_ticks() + 500; //ms
        led_on = !led_on;
        if(led_on)
            gpio_put(LED_PIN, 1);
        else
            gpio_put(LED_PIN, 0);
    }

    on_execute_realtime(state);
}

void my_plugin_init (void)
{
    // Add info about our plugin to the $I report.
    on_report_options = grbl.on_report_options;
    grbl.on_report_options = on_report_my_options;

    // Add blink LED function to grblHAL foreground process
    on_execute_realtime = grbl.on_execute_realtime;
    grbl.on_execute_realtime = blink_led;

    gpio_init(LED_PIN);
    gpio_set_dir(LED_PIN, GPIO_OUT);
}
```

Be aware that if the selected pin is assigned by the board map as an input it will not work as the driver pin configuration takes place after `my_plugini_init()` is called. If assigned as an output it might work...

Juank23, 2023-05-23:
I put an external led and set it on a pin that is free but still the same error. 
[image]

terjeio, 2023-05-23:
Save as _my_plugin.c_ in the same directory as _driver.c_ - do not modify the one in the _grbl_ folder.

Juank23, 2023-05-23:
in the grbl folder there is a file with the following name, copy a new file
where driver.c is located, my_plugin.c and the one that is in the grbl folder I left it blank but the same error continues, my question is that to activate my plugin is where it says OFF change it to ON or from where do I activate it?

Juank23, 2023-05-23:
Ready my friend, after several attempts and a couple of modifications it works now
 perfect, thanks for your help....

closed 2023-05-25 by Juank23