grblHAL/RP2040 #66: The question regarding the output pulse frequency of RP2040.

CHTUZKI, 2023-07-02:
HI.

I noticed that grblHAL supports many different MCUs, and I am curious if there are any differences in the output pulse frequency when driving motors with these MCUs, such as RP2040, which has PIO peripherals. Does it have a high frequency when driving motors? What is the maximum output frequency of grblHAL used to drive motors?

thanks~~

terjeio, 2023-07-02:
>  I am curious if there are any differences in the output pulse frequency

There are.

> Does it have a high frequency when driving motors?

Certainly higher that the 30 kHz limit of legacy Grbl.

>  What is the maximum output frequency of grblHAL used to drive motors?

It depends on several factors: processor speed, interrupt latency, number of axes, ... The iMXRT1062 (Teensy 4.x) driver is the fastest, 300KHz+ is possible, one person [pushed it to 600 kHz](https://www.youtube.com/watch?v=2fAEQEkQyFw) with modified code.

CHTUZKI, 2023-07-02:
Thank you for your reply.

I believe you have heard of LinuxCNC. I am not very familiar with LinuxCNC, but I found on YouTube that LinuxCNC can communicate directly with etherCAT servo drives to control motors. I posted a question on the LinuxCNC forum and I really want to know the difference between using etherCAT and the traditional way of sending pulses through the PC parallel port. The answer I got from the forum is that if you use etherCAT, you don't have to worry about PC delay. LinuxCNC will send some coordinates and data to etherCAT for processing. I don't know if I understand it correctly, and I'm not sure about the logic behind it.

If that's the case, would adding an etherCAT plugin to grblHAL and communicating with etherCAT servo drives, sending the parsed G-code motion data to the servo drives, improve the output frequency or make the control better?

closed 2023-07-02 by CHTUZKI

terjeio, 2023-07-04:
> If that's the case, would adding an etherCAT plugin to grblHAL and communicating with etherCAT servo drives, sending the parsed G-code motion data to the servo drives, improve the output frequency or make the control better?

I do not know if it is possible/feasible without core changes. At least a driver has to be made in order to explore such a setup and I would not use a RP2040 as a test platform for doing that. A STM32F7xx or a STM32H7xx with ethernet is, IMO, better suited.

You have to define the aim "make the control better" before starting, better than what and better for what purpose? The iMXRT1062 (Teensy4.x) driver is the fastest MCU supported by grblHAL - it should give you step rates in excess of 300KHz with little jitter, far better than Mach3 or LinuxCNC with a parallel port interface? Not good enough?

CHTUZKI, 2023-07-04:
😁That's right, grblHAL is indeed very excellent, and I feel that pursuing output frequency excessively doesn't make much sense.