grblHAL/RP2040 #67: RGB plugin & WS2812

terjeio, 2023-07-20:
Good morning, I tell you that I changed the code, it compiles but does nothing else I remain attentive thanks
[image]

_Originally posted by @Juank23 in https://github.com/grblHAL/Plugin_fans/issues/2#issuecomment-1640958928_

Juank23, 2023-07-30:
Good morning, @Terjeio, I still haven't been able to get it to work.

terjeio, 2023-07-30:
See this [comment](https://github.com/grblHAL/Plugin_fans/issues/2#issuecomment-1643203401) - can you provide the code?

Juank23, 2023-08-06:
Good morning Terjeio, I had a little trouble, I downloaded the new code.... here I send you thank you
[RP2040N.zip](https://github.com/grblHAL/RP2040/files/12270068/RP2040N.zip)

Juank23, 2023-08-07:
Good morning Terjeio, I tested with a strip of rgb leds that only have one signal pin and it would be good to modify the plugin to work with those rgb leds because they only work with a signal pin and we save pins that can be used for other functions, also note that when adding the plugin fans to choose the 4 gives an error and only shows 2 in io sender configuration here I send you the code with the tests I'm doing because I plan to make a pcb that has several inputs and outputs for certain functions thanks ....
[RP2040N.zip](https://github.com/grblHAL/RP2040/files/12282837/RP2040N.zip)

terjeio, 2023-08-07:
I only need the _rgb.c_ file not the whole project...
The changes I suggested in [this comment](https://github.com/grblHAL/Plugin_fans/issues/2#issuecomment-1633775601) has not been made, did you zip with the wrong version?

> also note that when adding the plugin fans to choose the 4 gives an error

Compile time error? Which? With `#define FANS_ENABLE 4` in _my_machine.h_?

Juank23, 2023-08-07:
good day Terjeio, well doing the tests I did that with the plugin of the fans, but if I just leave it in 2 work well, turn on and off with the command, but the rgb does not work, I have worked in lathe mode and works super with the new iosender, excellent your work .... It would be nice to add a drilling profile program, I am also trying to make the #define SPINDLE_SYNC_ENABLE 1 work, although I have a servomotor that I got to change it for the one I currently have.
I remain attentive thank you very much 
[image]
[image]
[image]

terjeio, 2023-08-08:
What is your $pins output (for the aux out pins)?

>  It would be nice to add a drilling profile program

What is that supposed to do?

> I am also trying to make the #define SPINDLE_SYNC_ENABLE 1 work

Spindle sync is not implemented for the RP2040 and I have no plans for adding it since the processor does not have a floating point unit (FPU).

Juank23, 2023-08-13:
Good day Terjeio, the fans are already working .... and aux outputs, the rgb plugin still can not make it work ....

terjeio, 2023-08-13:
Again, what is your $pins output (for the aux out pins)?

Juank23, 2023-08-13:
Good morning Terjeio, this is the configuration that I have.
I remain attentive thank you
$pins
[PIN:10,Feed hold]
[PIN:24,Cycle start]
[PIN:22,Probe]
[PIN:26,X limit min]
[PIN:27,Y limit min]
[PIN:28,Z limit min]
[PIN:29,A limit min]
[PIN:16,Aux input 0,P0]
[PIN:17,Aux input 1,P1]
[PIN:18,Aux input 2,P2]
[PIN:0,X step]
[PIN:2,Y step]
[PIN:4,Z step]
[PIN:7,A step]
[PIN:1,X dir]
[PIN:3,Y dir]
[PIN:5,Z dir]
[PIN:8,A dir]
[PIN:9,Steppers enable]
[PIN:11,Spindle PWM]
[PIN:12,Spindle on]
[PIN:13,Spindle direction]
[PIN:15,Flood]
[PIN:14,Mist]
[PIN:19,Aux out 0,P0]
[PIN:20,Aux out 1,P1]
[PIN:21,Aux out 2,Fan 0]

terjeio, 2023-08-14:
I have modified and renamed rgb.c as my_plugin.c for you, it outputs pulses on pin 16 so it might work:

[my_plugin.zip](https://github.com/grblHAL/RP2040/files/12332755/my_plugin.zip)

I leave it to you to debug, if required, as I do not have a WS2812 at hand for testing. Configure as for the [first test I a made for you](https://github.com/grblHAL/Plugin_fans/issues/2#issuecomment-1624772041).

Juank23, 2023-08-15:
Good evening terjeio, I tell you that I installed the prugin and it works fine but for some reason when I restore the driver with $rst=* it keeps thinking and it crashes I will check and let you know ....
thanks for your help ....

Juank23, 2024-03-28:
Good day terjeio, I downloaded your new code but I do not understand how this new rgb plugin works, how to configure a specific pin output led.... stay tuned thanks

terjeio, 2024-03-28:
Add to the map file:
`#define NEOPIXELS_PIN` pin_number

The RGB plugin uses [Marlin style M150](https://github.com/grblHAL/Templates/blob/5e2da4889b80b4274266bbd80c7d350ef8275e6b/my_plugin/RGB_LED_M150/my_plugin.c#L9-L21) command to control the LEDs.

Note that for the RP2040 a free PIO state machine is required which may not be available. And I just updated the plugin - I forgot to do that earlier - after a core change.

Juank23, 2024-03-29:
I am more confused, because I have already added #define NEOPIXELS_PIN 16 to the map
of my rp2040 zero but does not turn on the led, also note that it does not connect to other sender, such as candle ... I remain attentive thanks

terjeio, 2024-03-29:
M150 returns an error?

Juank23, 2024-03-29:
good morning terjeio, I do not understand what you mean rp2040 pio, and I get this error : error:20 - Unsupported or invalid g-code command found in block.

terjeio, 2024-03-30:
RGB plugin (_my_plugin.c_) is not compiled in? You have to [add it to the build](https://github.com/grblHAL/RP2040/blob/f2d2604564c14cb53c743fab2663bada9e169064/CMakeLists.txt#L12). If compiled in and showing up in the `$I` report then the driver code failed to add RGB support.

Juank23, 2024-03-30:
Good morning terjeio, I made it "Add my_plugin.c to build" OFF and 
"Add my_plugin.c to build" ON but it still doesn't work.
What do you mean by free PIO state machine???? stay tuned thanks

terjeio, 2024-03-31:
PIO state machines are used to handle step timings etc. and there are eight of them. Driving neopixels requires one and if none is available the driver code is not activated. The number of state machines used is dependent on board configuration, e.g. the SKR Pico configuration uses one for each axis and may thus run out of free state machines leaving none for the neopixel code.

Does the RGB plugin show up in the `$I` report?
Does M150 still report an error?

Juank23, 2024-03-31:
shows nothing $I...
shows nothing in M150
stay tuned thank you

terjeio, 2024-04-01:
The controller is dead and does not respond to any commands?

Juank23, 2024-04-02:
the controller responds with the $i command, it shows the firmware version grbl etc, but with the M150 command it shows nothing, it connects well with iosender, but does not connect with candle and lasergrbl.... thanks for your attention

terjeio, 2024-04-02:
If M150 does not respond with anything then your controller has crashed? No commands are accepted after it?

Juank23, 2024-04-02:
Good morning, it does not show anything with the M150 command, it does not turn on the rgb led.... but the controller works fine in iosender moves the motors and in $i shows all the firmware information, does not show anything of the rgb plugin as if it were not attentive.

terjeio, 2024-04-02:
> it does not show anything with the M150 command

M150 should generate an error if the plugin is not activated - here is my `$I` output, the last line is the RGB plugin:

```
[VER:1.1f.20240330:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[FREE MEMORY:219K]
[DRIVER:RP2040]
[DRIVER VERSION:240330]
[DRIVER OPTIONS:SDK_1.5.1]
[BOARD:PicoCNC]
[COMPATIBILITY LEVEL:2]
[AUX IO:5,8,0,1]
[PLUGIN:RGB LED (M150) v0.03]
```
Note that the RGB plugin is a [new one](https://github.com/grblHAL/Templates/blob/5e2da4889b80b4274266bbd80c7d350ef8275e6b/my_plugin/RGB_LED_M150/my_plugin.c#L9-L21), and it has a setting for the LED strip length - $536. This should default to 1 - check that it is not set to 0.

If a valid M150 command is issued, e.g. `M150B100` and no error is generated the LEDs should lit up blue. If not then the pin is not connected to the LEDs or it/they are not WS2812 or WS2812 compatible?

Why Candle and Lasergrbl does not comuunicate could be due to not setting the DTR handshake signal correctly on connect. Both communicate with some other grblHAL controllers such as STM32 Nucleo based or ESP32, but not the Teensy 4 (iMXRT1062). Differences in the USB stack?

Juank23, 2024-04-03:
for some reason it was at zero, now the LED is blue,
but I do not understand how are the functions I remain attentive thank you

terjeio, 2024-04-03:
> but I do not understand how are the functions I remain attentive thank you

You mean the words/parameters that can be used by [M150](https://marlinfw.org/docs/gcode/M150.html)? Or how the [machine state can be output to the LEDs](https://github.com/5ocworkshop/grblhal-rgb-plugin/pull/6)?

Juank23, 2024-04-03:
good morning tejeio... yes sir, to send the status of the machine to the leds....

terjeio, 2024-04-04:
... > send the status of the machine to the leds....

Follow the second link in my previous comment to find one implementation. `hal.rgb` has to be changed to `hal.rgb0` to make it work though.

You may also write your own plugin that uses the [hal.rgb0](http://svn.io-engineering.com/grblHAL/html/structrgb__ptr__t.html) interface to control the LEDs.

Juank23, 2024-04-05:
I have searched for hal.rgb, but did not find any information, thank you.

terjeio, 2024-04-05:
Here are a [few instances](https://github.com/5ocworkshop/grblhal-rgb-plugin/blob/409cbd04f61f1e9e1d7db508b671af6323d73cc5/rgb.c#L639).

Juank23, 2024-04-05:
hummm, now I'm more confused, I see this plugin more complicated than the previous one, thanks for your help ...

terjeio, 2026-04-01:
> terjeio, Could this work???

You will have to try to find out. If `M150R100` is not working then likely not.

terjeio, 2026-04-05:
> terjeio, Could this work???

Not without changes - it is outdated. Use the [current version](https://github.com/5ocworkshop/grblhal-rgb-plugin/blob/main/status_light.c) or update the one you have to match. If you use the current one comment out or remove the bit_count() function, rename status_light_init to my_plugin_init, add #define STATUS_LIGHT_ENABLE 1 to my_machine.h and set the AddMyPlugin option to ON in CMakeLists.txt.