grblHAL/RP2040 #70: IRQ_I2C_Strobe can't claim when SPI_IRQ_BIT is set.

andrewmarles, 2023-08-09:
Little issue here when trying to use both the keypad and W5500:

https://github.com/grblHAL/RP2040/blob/fe82ddb967c682dc549bd28dfe984241e3ced548/driver.c#L558

When both #defines are set, 'ok' gets reset to 0 when trying to claim the I2C interrupt.  I fixed it with this mod:

```
#if I2C_STROBE_BIT || SPI_IRQ_BIT

static bool irq_claim (irq_type_t irq, uint_fast8_t id, irq_callback_ptr handler)
{
    bool ok;

#if I2C_STROBE_BIT
    if((ok = irq == IRQ_I2C_Strobe && i2c_strobe.callback == NULL)){
        i2c_strobe.callback = handler;
        return ok;
    }
#endif

#if SPI_IRQ_BIT
    if((ok = irq == IRQ_SPI && spi_irq.callback == NULL)){
        spi_irq.callback = handler;
        return ok;
    }
#endif

    return ok;
}

#endif
```

terjeio, 2023-08-10:
I have applied the same fix as for the STM32F4xx.

andrewmarles, 2023-08-10:
Thought it sounded familiar.

closed 2023-08-10 by andrewmarles