grblHAL/RP2040 #71: Pins Probe and homing

Juank23, 2023-08-13:
good morning Terjeio, to save pins on my board and because of the shape of my machine it is very easy to use the probe pin with the limit X pin, I marked both pins with the same number, i.e. ,

#define LIMIT_PORT GPIO_IN
#define X_LIMIT_PIN 26

#define PROBE_PIN 26

homing works but the probe stops working, it reaches the sensor but does not return it is stuck, but if I change the pin number of the homing pin it works fine, is there a way to do these functions well with a single pin....

terjeio, 2023-08-13:
Hard limits is enabled? It may work without.
A plugin that intercepts probe and limit calls might work around any issues, disable X limit IRQ when probing and restore it when completed.

andrewmarles, 2023-08-13:
As a possible example, the probe protection plug-in can enable hard limits when probing (people connect overtravel to z axis limit).

https://github.com/Expatria-Technologies/grblhal_probe_plugin/blob/31182e0536b654c7822ff17c9ffd41676f25827e/probe_plugin.c#L240

latest core probe events are very flexible.

Juank23, 2023-08-13:
good day andrewmarles, I must add the plug-in for the function I need? this plug-in works with the pin number that I have on my map.... stay tuned thanks

Juank23, 2023-08-13:
Good morning Terjeio, if the stroke limits are activated for the home I remain attentive thanks

terjeio, 2023-08-13:
There is no need to enable hard limits ($21 > 0) for homing to work. And if the machine is homed soft limits ($20=1) will take care of keeping motion within limits if maximum travel is correctly set up.

Juank23, 2023-08-13:
Terjeio, the problem is that by sharing the same pin number probe and x_limit the probe stops working but the home function if it does well I remain attentive thank you ....
[image]

terjeio, 2023-08-13:
Is the limit switch normally closed (NC) and wired in parallel with the probe switch? If so you have to use a normally open (NO) switch, and the same for the probe, to make it work. Alternatively both have to be NC and wired in series.

Juank23, 2023-08-13:
terjeio, simply by sharing the same pin, the probe routine stops working.

#define X_LIMIT_PIN 26

#define PROBE_PIN 26

Is there any solution or is it necessary to use a different pin for each function? thanks for your attention

terjeio, 2023-08-13:
> Is there any solution

[This](https://github.com/grblHAL/RP2040/issues/71#issuecomment-1676455227)? How are you wiring the limit switch and probe and what kind of switches are they (NO vs NC)? If the limit switch is wired in parallel with the probe (switch)  and of the NC variety then the probe signal will be shorted out and will never be seen by the processor. And vice versa...

Juank23, 2023-08-13:
i am using a hall sensor a3144, but as soon as i share the pin it stops working... without activating the home i stay tuned
[image]
[image]

terjeio, 2023-08-14:
I have verified that both X-limit and probe inputs can share the same pin as both shows up in the status report in the correct way.

From the screenshots above:

1. You have to disable hard limits, either permanently or by writing a plugin that disables X limit IRQ while probing. Instead of disabling the IRQ you can intercept the limits callback and block it from reaching the core when the X limit is triggered during probing. [This plugin](https://github.com/grblHAL/ESP32/blob/master/main/BlackBoxX32.c) can be used as a starting point for that as it intercepts both homing and limits calls and modifies the callback result depending on which axis/motor is homed. You'll have to intercept the [hal.probe.configure()](http://svn.io-engineering.com/grblHAL/html/hal_8h.html#abc871aa44f24e5ce6be547088ae48bb7) call instead of the homing calls and turn off the X limit trigger when the `probing` argument is true.
2. X limit and probe input needs to have the same inverted status, they do not now.
3. Since the X-limit "LED" is lit (red) and the probe "LED" is not the switch action might not be compatible. One is NC and one is NO and wired in parallel?. Both "LED"s should be lit/unlit in sync.
4. You have to have the same switch action for both the X-limit and the probe _and_ wire accordingly _or_ add some electronics to merge the signals. See [this post](https://openbuilds.com/threads/blackbox-w-support-for-3d-touch-probe-and-tool-height-sensor.20730/#post-137221) for ideas.

BTW your best/easiest alternative is perhaps to buy a Pi Pico or a Pi Pico W to get around the lack of pins. Alternatively I can design and test electronics and code for you. FYI my current rate as a consultant is EUR100/hr.

Juank23, 2023-08-14:
good day terjeio, the strange thing is that it does not work for me I have only one hall sensor sharing the two functions, moreover, without activating the homing as I show in the screenshots above it does not work it stays stuck in red led, I thought it was easier that issue.... on the other hand when I set the homing set machine origin 0 and activate the jog I have the limit_Y sensor in the middle of the axis
i.e. the axis travel is 80mm and is in 40mm then it takes from there the zero but when moving the axis with the limit jog only travels 40mm from zero ie the center of the axis, is there any way to configure that position to do all the travel I remain attentive thanks
[image]
[image]
[image]

terjeio, 2023-08-14:
> is there any way to configure that position to do all the travel

For non standard homing only via a plugin that hooks into `grbl.on_homing_completed()` and sets the needed parameters for the core by overriding those set by [limits_set_machine_positions()](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/machine_limits.c#L142-L164) and possibly [limits_set_work_envelope()](https://github.com/grblHAL/core/blob/e2df2fb101292a02065f913a909056522dfe16d9/machine_limits.c#L114-L138).

Juank23, 2023-08-26:
Good morning Terjeio, is there any way to configure these values for each independent axis, I remain attentive thank you.

terjeio, 2023-08-26:
>  is there any way to configure these values for each independent axis

There are [10 preallocated setting numbers](https://github.com/grblHAL/core/blob/1d7872fc5cd08da3b79e252ddb0fa0e5986829f7/settings.h#L278-L288) for [user defined settings](https://github.com/grblHAL/Templates/blob/master/my_plugin/settings/my_plugin.c) so I guess yes.