grblHAL/RP2040 #73: Support for TMC5160

fireup924, 2023-08-31:
I would like to add support for TMC5160 which uses SPI but not sure where to start?

Do I add a new file call tmc_spi.c and include the two functions:

tmc_spi_write() and tmc_spi_read() ?

terjeio, 2023-08-31:
It is a little more involved than that, best is to start with code from another driver. There are two SPI modes for Trinamic drivers, either separate chip select lines for each drivers or a common one. The latter requires the SPI lines to be chained and the drivers appear as a single long register.

Part of the [STM32F4xx driver code ](https://github.com/grblHAL/STM32F4xx/blob/master/Src/tmc_spi.c)can be used as a starting point. To save pins you can use the existing SPI interface provided in [spi.c](https://github.com/grblHAL/RP2040/blob/master/spi.c).

terjeio, 2024-05-09:
The STM32F4xx driver code linked above has separate chip select lines for each driver, this[ board specific code](https://github.com/grblHAL/STM32F4xx/blob/master/boards/st_morpho.c) has a shared line for chained drivers.

_driver.c_ needs to be updated to handle the chip select(s), again see the STM32F4xx driver [here](https://github.com/grblHAL/STM32F4xx/blob/60c0d3f2061a27d5585735a709a63df6759efaa2/Src/driver.c#L375-L395) and [here](https://github.com/grblHAL/STM32F4xx/blob/60c0d3f2061a27d5585735a709a63df6759efaa2/Src/driver.c#L2321-L2325). The latter code ensures the pin(s) are enabled as outputs with the level set high initially.

fireup924, 2024-05-11:
Thank you for the help. I got it working now. I'm using the output from HC595 shift register for CS pins and I didn't initialize them as high and was causing SPI bus collision