grblHAL/RP2040 #74: BTT Pico UART addressing bits transposed

Barney-F, 2023-09-05:
There seems to be a bit of a miss-match when connecting to the BTTpico's onboard TMC2209 stepper drivers via UART. 
In IOsender's 'trinamic tuner' tab the data for 'Y' and 'Z' appear to be transposed, both for the 'get status, all drivers' and also when graphing the SGRESULT. 
The published schematic shows the following addressing for the MS1_AD0 and MS2_AD1 pins on the TMC2209s:

x: AD0=0, AD1=0
Y: AD0=0, AD1=1
Z: AD0=1, AD1=0
E: AD0=1, AD1=1

I'm guessing the '01' and '10' might be the wrong way around. 
I've had a look at the code, looking for where this might be corrected, and found Plugins_motor/trinamic.h : L.261 trinamic_driver_config_t.
https://github.com/grblHAL/Plugins_motor/blob/11c937c85f07f7f4a52be6dfdd0e49f4fcfe1886/trinamic.h#L261
The comment notes this can be overridden by the driver interface. 
Is this something that can be defined in the board_map.h?

terjeio, 2023-09-13:
Are you compiling locally or with the Web Builder?
I do not have access to a SKR Pico board so cannot verify code changes.

If you compile locally replace [this code](https://github.com/grblHAL/RP2040/blob/7467dbf5e33bfbe393e59e6d6610ba4690d136aa/tmc_uart.c#L83-L89) with:
```
#if defined(BOARD_BTT_SKR_PICO_10)

void driver_preinit (motor_map_t motor, trinamic_driver_config_t *config)
{
    static const uint8_t address_map[4] = { 0, 2, 1, 3 };
    
    config->address = address_map[config->address];
}

void tmc_uart_init (void)
{
    static trinamic_driver_if_t driver_if = {
        .on_driver_preinit = driver_preinit
    };

    memcpy(&tmc_uart, serial2Init(230400), sizeof(io_stream_t));

    tmc_uart.disable_rx(true);
    tmc_uart.set_enqueue_rt_handler(stream_buffer_all);

    trinamic_if_init(&driver_if);
}

#else
    
void tmc_uart_init (void)
{
    memcpy(&tmc_uart, serial2Init(230400), sizeof(io_stream_t));

    tmc_uart.disable_rx(true);
    tmc_uart.set_enqueue_rt_handler(stream_buffer_all);
}

#endif // BOARD_BTT_SKR_PICO_10
```

terjeio, 2023-09-15:
> I've noticed another issue with the BTT SKR-pico board concerning the timing of the TMC2209 enable pins being asserted before the motor current is configured.

What exactly is the issue with this?

Barney-F, 2023-09-15:
When power is first applied to the board, it draws lots of current and forces my PSU to over-current hiccup. 
After about 3 seconds, the board current drops down to what should be right for the configuration. 
My guess is that the enable pins are asserted before the RP2040 has had a chance to set the correct current via UART, and the TMC2209's push maximum current through the motors until this is properly setup.  
I was going to tack solder some wires onto the UART and enable pins and hook them up to a logic analyzer in order to check that this is actually what's happening. 
The TMC2209's have some onboard OTP (One Time Programmable) configuration bits that BTT may have  pre-programmed in order to force this board into a working state without having to get the UART communications functioning, perhaps this is what I'm seeing.

Barney-F, 2023-09-15:
"lots of current" on start-up is roughly 8A (hard to measure when its hiccuping).
My idling current is about 300ma@24v , as I'm running 4 x Nema11 steppers at modest current levels

terjeio, 2023-09-15:
Do you have $1=255 (steppers always on) or $37 non-zero? These are found in the _Stepper_ group.

Barney-F, 2023-09-15:
I have $1 = 255 and $37 = 7

terjeio, 2023-09-15:
Ok, what happens when you set $1 = 25 and $37 = 0?

Barney-F, 2023-09-15:
With $1=25 and $37 = 0 :  The startup current peak is now about 200ma, and fades off to an idling current of ~35ma. 
(jogging is a bit jankey with the steppers energizing at the start of each move)

terjeio, 2023-09-15:
Ok. IIRC the drivers I have initializes with a low current setting so it might well be that OTP bits has been changed.
Anyway, how to handle this? A test could be to change the init sequence to see if that helps. Currently the stepper drivers are initialized post driver setup [here](https://github.com/grblHAL/Plugins_motor/blob/11c937c85f07f7f4a52be6dfdd0e49f4fcfe1886/trinamic.c#L1799-L1809). Can you try to change that to pre driver setup like this:

```
static bool on_driver_setup (settings_t *settings)
{
    hal.delay_ms(100, NULL); // Allow time for drivers to boot
    trinamic_drivers_setup();

    return driver_setup(settings);
}
```

This may work with UART drivers but not SPI drivers since GPIO pins are not initialized pre `driver_setup()` but UARTs are.

Barney-F, 2023-09-15:
That test works great :
Startup current is about 1.8A, and then stabilizes to ~300ma
My PSU is much happier

Barney-F, 2023-09-16:
So after some digging in the TMC2209 datasheet, there's a note on pg 52 of rev1.09 that cautions the start-up current.
The note is related to sizing the current sense resistors - but it's still relevant. 
Connection to the TMC2209 Vref pin also influences the max current (not connected in the BTT SKR-pico, so defaults to 73% pg.53)   usually connected to a trimpot on a 'stepstick' style module, so this might be where your test setup differs to a SKR-pico.
Without having a BTT SKR-pico board on the bench in front of you, this wouldn't be in any way obvious.
Thanks again for your time looking into this issue.

[image]

closed 2023-11-05 by Barney-F