grblHAL/RP2040 #75: ATC using Macros

fireup924, 2023-09-26:
I notice there's some code in the macros.c that has ATC code that uses macros to preform automatic tool changes. I want to activate this feature when there's am M6 command and it runs the tc.macro that is stored in the littleFS or SD card. Which flag to I have to set to turn this feature on?

terjeio, 2023-09-26:
[SDCARD_ENABLE](https://github.com/grblHAL/RP2040/blob/aedc240f0bd71a12d0efbbf4b32732d9a2fdf3c4/my_machine.h#L51)
[NGC_EXPRESSIONS_ENABLE](https://github.com/grblHAL/core/blob/70ffca3c0afcddcbf92be837b47cde83295fc979/config.h#L510)
Using LittleFS is a little complicated for the time beeing, you will need a Pico W and enable WebUI or WiFi with FTP.
Upload is then either via the [WebUI maintenance](https://github.com/grblHAL/Plugin_WebUI/tree/9dbc8ce12cdb55e1e4add4cc04e71b34087744f7#installation) page or FTP.

Uploads to the SD card is possible via [ioSender](https://github.com/terjeio/ioSender) if `SDCARD_ENABLE` is set to `2` (by using a slightly modified YModem protocol).

terjeio, 2023-09-26:
More [here](https://github.com/grblHAL/core/wiki/Expressions-and-flow-control#discussion).