grblHAL/RP2040 #77: Keyboard Jogging does not work when using Ethernet-to-uart module

fireup924, 2023-10-26:
When using IOSender and communicating via [Ethernet-Uart module](https://www.waveshare.com/uart-to-eth-b.htm), the Keyboard jog does not work and gives a 
`"G-code words consist of a letter and a value. Letter was not found."` error.

I traced it down to this section of protocol.c on line 890 `hal.stream.cancel_read_buffer();` commenting this line out and adding `system_set_exec_state_flag(EXEC_MOTION_CANCEL);` seem to fix the problem.
```
case CMD_JOG_CANCEL:
  char_counter = 0;
  drop = true;
  //hal.stream.cancel_read_buffer();
  if (state_get() & STATE_JOG) // Block all other states from invoking motion cancel.
       system_set_exec_state_flag(EXEC_MOTION_CANCEL);
  if(grbl.on_jog_cancel)
      grbl.on_jog_cancel(state_get());
  break;
```
I'm not sure if this is the right fix for this? I seems the keyboard jog command for IOSender is sending a 0x85 and a [$J=](url) in the same line and cause the hal.stream.cancel_read_buffer() to trim some of the Jog string...

fireup924, 2023-10-26:
Thank you for the fix. It works now without having to touch the core code in protocol.c

terjeio, 2023-10-27:
>  It works now without having to touch the core code in protocol.c

I spent a lot of time fixing jog cancel that intermittently failed with the legacy code inherited from Grbl, so I suspected the issue was somewhere else. What is odd is that somehow spurious characters get inserted in the input buffer when attempting to flush the RX FIFO. The solution was not to flush it since characters received after the jog cancel command are not problematic and should be kept anyway...

closed 2023-10-27 by terjeio