grblHAL/RP2040 #86: Enable Macros cause crash

fireup924, 2024-04-10:
When I enable SDCard/Macros, ioSender would not run and locks up.

When I connect to a terminal program and send a $I, it will cause the board to lock up.

terjeio, 2024-04-10:
Does not happen for me so need more details.

terjeio, 2024-04-10:
Which board?
Is it ethernet enabled - but grblHAL compiled without ethernet support?
Asking since mounting the SD card fails for me when there is an ethernet chip on the SPI bus.

> Do I have to format the card a certain way? or use a certain card?

Fat32 formatted, size <= 32GB. Some makes/models of SD cards may fail to mount.

fireup924, 2024-04-10:
The board does not have pull-up resistors but I'm using it with this SD Card module that has pull-up resistors:

[image]

fireup924, 2024-04-10:
I will try a Web Build with just a bare Pi Pico module and this SD Module and see if it works on that...

fireup924, 2024-05-09:
Turn out my SD Card was not formatted correctly.

closed 2024-05-09 by fireup924