grblHAL/RP2040 #91: High motor current on BTT SKR Pico 1.0, No communication with Trinamic drivers

richardklingler, 2024-06-18:
Good day (o;

Got me some BTT SKR Pico boards for testing several CNC setups.

Compiled in VSCode (git pull from today)  on macOS as the web builder doesn't work on macOS Safari and some settings can't be changed there (works in Firefox).

When moving a motor the current for an axis goes up high (> 2A).

M122 reports then:

[MSG:Warning: Could not communicate with stepper driver!]

After resetting the board, M122 reports the TMC2209 setting and motor current goes back to normal.

Anyone else seen this behavior?

terjeio, 2024-07-01:
Could be [too short delay](https://github.com/dresco/STM32H7xx/issues/29) before initializing the drivers after power up? If motor power is not up then communication with the drivers fails. If the controller is powered from a computer then this should be powered up after the motor power is switched on.

richardklingler, 2024-10-07:
Finally had some time today to test it...

Power supply from external, no USB power....

Two motors plugged into X and Y, power on -> high motor current....
Unplug motor X -> normal current....

Two motors plugged into Y and Z -> normal current.....

Even if I press reset button the high motor current stays with XY plugged in....also when controller is power from USB and power is first applied to motors and then to USB -> high current....

Suspect some hardware issues with the boards, and all 4 boards I have behave the same -> BTT SKR Pico = rubbish (o;

c---, 2024-10-20:
@richardklingler It wasn't that long ago that the drivers in here would make the motors "scream" in pain when I tried them on any board using the TMC2209. It's better now but then there are issues like this. I have not tested for this issue specifically yet.

I don't think it's the board but I guess it could be a quirk. I use many of them with various other firmware without issues. Though there was one other firmware that also had problems with the TMC2209 (the name of the firmware escapes me at the moment).

There don't seem to be many people testing grblHAL with these boards since most use Klipper.

cbaugher, 2024-11-11:
I just setup an SKR Pico on a small 3 axis engraving machine, and it's working fine.  I used the web builder with the latest grblhal as of maybe a week or so ago.  A fair number of the settings had to be adjusted because the defaults were wrong.  I can post my settings if it would help.