grblHAL/RP2040 #94: Compilation error if bluetooth is enabled in native mode

EKazakof, 2024-08-27:
If you enable bluetooth in native mode, the Web Builder and visual studio code generates a compilation error.

web builder output:

Command failed: make
/home/dev/RP2040/driver.c: In function 'cyw43_led_on':
/home/dev/RP2040/driver.c:2238:9: warning: implicit declaration of function 'cyw43_arch_gpio_put' [-Wimplicit-function-declaration]
2238 | cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, data != NULL);
| ^~~~~~~~~~~~~~~~~~~
In file included from /home/sdk/pico-sdk/src/rp2_common/pico_cyw43_arch/include/pico/cyw43_arch.h:16,
from /home/dev/RP2040/bluetooth.c:36:
/home/sdk/pico-sdk/lib/cyw43-driver/src/cyw43.h:40:10: fatal error: lwip/netif.h: No such file or directory
40 | #include "lwip/netif.h"
| ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/grblHAL.dir/build.make:468: CMakeFiles/grblHAL.dir/bluetooth.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1494: CMakeFiles/grblHAL.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

terjeio, 2024-08-27:
Web Builder should be ok now, VSCode may still fail (Linux only?) due to a name clash - will commit a fix for this later.

EKazakof, 2024-08-28:
Thank you for a quick response. VS Code for Windows Pico-SDK v1.5.1 compilation is successful.

closed 2024-08-28 by EKazakof

marcosgabrielserrano, 2024-10-20:
Is there going to be a fix for the 2.0 sdk?

terjeio, 2024-10-20:
> Is there going to be a fix for the 2.0 sdk?

Yes - when I can get VSCode to compile with it. Updating the SDK did not work for me.

terjeio, 2024-10-26:
@marcosgabrielserrano The current version seems to work with SDK 2.0.0, and it also builds for the RP2350 - but since I do not yet have access to a board I cannot test it.
If you have a specific issue with 2.0.0 please raise a new issue since this has been closed by the OP.

marcosgabrielserrano, 2024-11-16:
@terjeio 

It turns out that my sdk didn't fully upgrade when I set it to 2.0. After clearing completely and reinstalling it works.