grblHAL/RP2040 #99: Laser mode turns off spindle when g1 over previous path

MadTooler, 2024-11-27:
Long time no issue.  Thanks for all the progress since last!

While running in laser mode and under feed with G01, the spindle is disabled when the next G01 travels over the same path as last.  I am not sure if it actually cycles super quickly as it returns, or just disables.  Either way, it is not the expected.  

For example:

M4 S1
G01 X5 Y0 F5000
G01 X-5 Y0  (this will result in the spindle being cycled at its start)
G01 X1 Y0  (this will also result in the spindle being cycled at its start)

I am running a 1.1f version of grblHAL/RP2040 from last year.  I did not see this issue noted or referenced in the changes.  If I missed it, let me know and I will update my firmware to any corrections you may have already made.

Thanks again!!

MadTooler, 2024-11-29:
... furthermore, I have noticed some gcode runs fine, then, another work offset setup with the same code has the issue.  I bet the minor zero position offset variation, due to rounding, which sometimes shows a slightly not zero in some senders and with some mechanical setups, may be the culprit. 

I am curious if this is something that needs an adjustment to code, or if there is a setting somewhere that can make this an easy tweak.

terjeio, 2024-11-30:
Please update and try again as I cannot see any incorrect behaviour.

MadTooler, 2024-11-30:
Thanks for looking into this.

I do believe it is quickly cycling spindle off then on again when ends the initial path and begins the next returning path that goes directly over the initial.  I would not expect to actually be aware of that with a laser because it is virtually instantaneous with that type of "spindle."  I only am aware because my "spindle" does not accept the back on command since it is still winding down from the off and ignoring new commands until it finishes that task.  

Do you know if you have changed any of the function of the laser mode in the last year that may be related?  If not, I do not expect any difference with an update.  Even so, I should update regardless, as soon as I double check what custom stuff I made before.

Related but off issue... I see that you have an embroidery plugin. I am running inkstitch within inkscape.  If you have not checked that out, it is pretty slick.  It takes a bit to get the work flow, but definitely worth it to be able to make anything you want and open source.

FYI the machine that I am using the laser mode on is also sewing related.  It is a longarm quilting machine.  Think of something like an embroidery machine but 12 foot or 4 meters long.  That means 2 axis cnc driven with the spindle being the running/stitching of the sewing machine itself.  I have been working on a new motion controller and the full interface that has features I have not seen in existing commercial products.  My goal is to make the software and most, if not all mechanical/electrical, open source while selling the kits.  Your grblHAL has worked very well for my application.  Thanks again!

terjeio, 2024-12-03:
> I do believe it is quickly cycling spindle off then on again when ends the initial path and begins the next returning path

The spindle off/on signal is not cycled, only the PWM signal is. If it does either $9 is set to RMP controls spindle on/off or there is a bug in your version.

> Do you know if you have changed any of the function of the laser mode in the last year that may be related? 

I have changed the spindle code quite a bit over the last year and I am not going back to an earlier version to debug this issue. If it is still there in the latest version I will look into it.

I know about inkstitch, sadly the old sewing machine is not working proberly so I cannot use it for anything bu crude testing.

>  I have been working on a new motion controller and the full interface that has features I have not seen in existing commercial products.

This is interesting, will you publish anything about this work?

MadTooler, 2024-12-03:
$9 is set to 3 per your direction here (https://github.com/grblHAL/RP2040/issues/47) with only a single pin for spindle on/off

I definitely do not expect any debug of an older version considering your further development.

Digging through to see if my tweaks may have impacted laser mode, I find I need to really rewrite a lot of my motor driver and possibly tweak GPIOs of grblHAL to match the revised approach my overall system is now taking.  I will make do for now in limp mode until I have a clean slate with everything updated.

I do plan to publish as soon as I get the major software and mechanical rework done (maybe v4.0 now?).  Ideally that would have been last year, but stuff always gets fun with a project this ambitious.  My control board, although fairly basic, is rather custom.  Mechanics are very custom.  It is not at a stage yet where I can get anyone else hands on unless they happen to be around the corner.  If you decide to take a trip to middle US, you are welcome to stop by :)

The current system 
- rpi4b running Node-RED with its dashboard v1 as GUI and as a custom gcode sender
- grblHAL on an rp2040 
- additional rp2040 working as a closed loop and other integrated special featured motor driver
- both features to allow for "free motion" manual quilting and full CNC (or what that industry refers to as "robotic")
- Intended for both modern machines and as a retro-fit to vintage longarms
- quilting paths are SVG based which means open source free design software like inkscape are available for path creation and sharing

MadTooler, 2024-12-12:
> And you are sure the same could not have been done via a plugin and/or custom board code?

I have mostly applied as plugin and custom board.  Overall, my changes are very slight and minor.  The bulk of the code I need to update is with the other parts of my system interacting with the grblHAL pico board.  I will need to revise the plugin and board code in grblHAL, but I have an overall system overhaul that is underway and will change what I actually will need to change in grblHAL.  

I can make do with the known quarks of the existing system as I run a few projects through the machine before diving in again and working through the yet to be known issues I may introduce with the wild new version I will be attempting.

As soon as it becomes possible, I will try your current version to see if the issue still occurs.  Thanks!