grblHAL/RP2040 #102: Setting `N_AXIS` in my_machine.h doesn't work properly

qwertychouskie, 2024-12-17:
When setting `N_AXIS` in `my_machine.h`, there are various compiler warnings about `N_AXIS` being redefined, and the resulting build only seems to support 3 axes.  If I change the default `N_AXIS` definition in `config.h` (line 42), the resulting build supports the expected number of axes.

Is there a different way `N_AXIS` is supposed to be defined?

terjeio, 2024-12-17:
> Is there a different way N_AXIS is supposed to be defined?

It has to be changed in _config.h_, from the compiler command line (e.g. `-D N_AXIS=4`) or for those who use PlatformIO - in _platformio.ini_.
The core does not use _my_machine.h_, this is a design choice made to keep the core isolated from driver implementations.

qwertychouskie, 2024-12-17:
If `my_machine.h` is not the correct place to set `N_AXIS`, it should probably be documented, ideally with a compiler error to catch the invalid configuration.  Getting grblHAL up and running took much longer than it would have if this issue were documented.

Another big issue I ran into that took a long time to figure out was the `*_DIRECTION_PIN` definitions being silently ignored.  Turned out the example config I copied from (`generic_map_4axis.h`) set `DIRECTION_OUTMODE` to `GPIO_SHIFT6`, resulting in the `*_DIRECTION_PIN` definitions being silently ignored.  Defining `*_DIRECTION_PIN` while `DIRECTION_OUTMODE` is not `GPIO_MAP` should probably at least be a compile-time warning, ideally an error.

terjeio, 2024-12-18:
>  Getting grblHAL up and running took much longer than it would have if this issue were documented.

Perhaps a clue is that `N_AXIS` and other config options from _grbl/config.h_ are not present in _my_machine.h_? And that _grbl/config.h_ even exists?

> Defining *_DIRECTION_PIN while DIRECTION_OUTMODE is not GPIO_MAP should probably at least be a compile-time warning, ideally an error.

There are limits to what can be catched compile time, it might not be possible or may take a lot of time to add and debug. So not a priority, at least not for me.

Ideally there should be a large team behind grblHAL, writing perfect code and add all sorts of documentation... The few that are working on grblHAL do it for free in their spare time, so sadly some that wants to use the code may waste their precious time on stuff like this issue. I am sorry for that.