grblHAL/RP2040 #103: FTP Compatibility with LittleFS

engigeer, 2024-12-18:
I am trying to compile firmware which allows upload of files to littleFS via FTP and have encountered two issues:

1. FTP seems to require the SD card plugin, which in turn requires definition of an SD_CS_PIN. For testing purposes, I have defined this with a spare pin, but for the final application I will not have any spare pins. I've seen some 'dummy' pin assignments to 0xFF elsewhere in the code, but in this case generates a compile warning. I'm wondering if there is a best practice for enabling features that require pin definitions without using up physical pins (as an example of another use case, I am hoping to control my coolant device with a custom Modbus plugin, and would like to enable coolant without a physical COOLANT_FLOOD_PIN).

2. Compiling with FTP_ENABLE=1 and SDCARD_ENABLE=1, I am able to see the settings for FTP in ioSender, but have not been successful in connecting / transferring files with WinSCP. After attempting to connect, winSCP generates an error "Could not retrieve directory listing. Requested action aborted: local error in processing." I am able to successfully ping the device over the network, as well upload files via the webUI.

terjeio, 2024-12-18:
There are several factors complicating this, and I am not sure how to "fix" them.
Since littlefs was originally added to support the WebUI there are several hard coded checks that has to be changed:

In _networking.c_ change [this line](https://github.com/grblHAL/Plugin_networking/blob/0203465c9c6a53ef187ad8a76ad27a48135353ad/networking.c#L65) to `#if FTP_ENABLE && (SDCARD_ENABLE || LITTLEFS_ENABLE)`

In _ftpd.c_ comment out or delete[ these lines](https://github.com/grblHAL/Plugin_networking/blob/0203465c9c6a53ef187ad8a76ad27a48135353ad/ftpd.c#L1171-L1173) and [this line](https://github.com/grblHAL/Plugin_networking/blob/0203465c9c6a53ef187ad8a76ad27a48135353ad/ftpd.c#L1279).

The first block of lines checks if there is gcode beeing streamed from the sdcard and blocks access if so. This should be changed to check for gcode beeing streamed from any local file system.
The second line is for mounting the SD card, this can be removed but there should be a check for filesystem availability, perhaps by mounting a null filesystem if SD card nor littlefs is available. Will have to think about this.

In _driver_opts.h_ change [this line](https://github.com/grblHAL/core/blob/50a7507f6fe98fa22c062685f46e9d06d20e85f8/driver_opts.h#L534C1-L534C21) to: `#elif !(SDCARD_ENABLE || LITTLEFS_ENABLE)`

These changes should make FTP available.

Next is how and when to mount the littlefs filing system. The RP2040 driver does not mount it unless the WebUI is enabled, [this line](https://github.com/grblHAL/RP2040/blob/b609fad6a82d0ab177e17f1b58e75de863db0601/driver.c#L2571) has to be changed to:` #if LITTLEFS_ENABLE` to fix that.

The next line is for mounting and where to mount littlefs, if left like this you have to specify which remote directory to mount in WinSCP like this:

[image]

Attempting to mount the root directory will fail, this is to be considered as a bug in grblHAL that should be fixed.

If only the littlefs filesystem is mounted then it would be better to mount it as root by changing the line to:
`    fs_littlefs_mount("/", pico_littlefs_hal());`
This could be done when the SD card plugin is not enabled or by defining `#define LITTLEFS_ENABLE 2` instead of `1` in _my_machine.h_ and check for that when mounting. I think I'll prefer the latter since there is code that searches for files in the _/littlefs_ directory  before checking for them in the root directory. This may have to be changed if littlefs is mounted as root. Again I'll have to think about this too, and I have to find out why mounting littlefs as root fails with ftp.

Anyway, these changes may get you going...
A bit more info about your use-case would be helpful in deciding how to change the code to allow littlefs to be used standalone.

terjeio, 2024-12-18:
WinSCP failing to read root when littlefs is mounted as root can be hacked to work by changing [this line](https://github.com/grblHAL/Plugin_SD_card/blob/8bb3c98566beaa337d01022b7a1cae0df4f2d883/fs_littlefs.c#L181) to `return *path == '/' ? 0 : -1;`
I'll add a better fix in the next commit. You may also consider adding `#define FF_FS_RPATH 1` to _my_machine.h_, but be aware I have not tested this and it may have side-effects.

engigeer, 2024-12-19:
Thank-you very much for the assistance. Following the guidance above I have successfully got FTP working. 

I think there are probably still a few code guards I need to update in order to use ENABLE_LITTLEFS=2 and prevent conflicts but otherwise all is working as expected.

For context, I am testing this with a picoHAL board (designed by the great folks over at Expatria). My plan is to bind Aux inputs to G65 PXXX commands (using the macros plugin), and then upload subroutines as PXXX.macro files via FTP. Target application is not necessarily to run a full CNC, but other smaller automation projects with just 1-2 motion axes and a few IO points. This approach would mean the automation sequence is just ‘programmed’ in g-code without the need for custom C or micropython firmware. I think very similar to the way that some folks in the DIY CNC community are using macros with ATCs.

engigeer, 2024-12-22:
I compiled test firmware for my device based on your most recent commit. All seems to be working as expected, but I did get a compile warning regarding a missing definition of *stream_is_file()* in ```networking/ftpd.c```. This seems to be resolved with the addition of ```#include "grbl/stream_file.h"``` at the top of the file. Thank-you again for your assistance.

terjeio, 2024-12-22:
> Target application is not necessarily to run a full CNC, but other smaller automation projects with just 1-2 motion axes and a few IO points. This approach would mean the automation sequence is just ‘programmed’ in g-code without the need for custom C or micropython firmware.

Interesting, You could publish it when ready, e.g. on Hackaday?

> ... I did get a compile warning regarding a missing definition of stream_is_file() in networking/ftpd.c

This will be fixed in the next commit.

engigeer, 2024-12-29:
> Interesting, You could publish it when ready, e.g. on Hackaday?
> 
> >

Yes, I hope to do so!

One of the first projects I am planning to use this for is an automated card-shuffling machine, which is actually inspired by a Hackaday article from a few years ago.

Thanks again for the commit, all is working well.

closed 2024-12-29 by engigeer