grblHAL/RP2040 #105: No pin affected to spindle when using GPIO

setaperlacloche, 2025-01-01:
On a regular basis I rebuild a RP2040 firmware from sources for my CNC with a custom board.
My previous build from 20240928 worked perfectly, but with 20241230 my spindle (a basic on/off spindle) refuses to start. Reason : no PIN affected to 'Spindle on' function.

**How to reproduce :**
The simplest way to reproduce this behavior is to used Web builder with Driver = RP2040, Board = Generic and Spindle 1 = On/Off.
==> Web builder configuration file :
`{"driver":"RP2040","URL":"https://github.com/grblHAL/RP2040","board":"","symbols":["OVERRIDE_MY_MACHINE=1","USB_SERIAL_CDC=1","PROBE_ENABLE=1","ESTOP_ENABLE=1","SPINDLE0_ENABLE=7","N_SPINDLE=1"],"docker_instance":"/RP2040","grblHAL":"webBuilder"}`
==> `$I` output :
```
[VER:1.1f.20241230:]
[OPT:NMDSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,SED,RTC]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH 4K]
[FREE MEMORY:212K]
[DRIVER:RP2040]
[DRIVER VERSION:241219]
[DRIVER OPTIONS:SDK_2.1.0]
[AUX IO:3,4,0,0]
[PLUGIN:FS macro plugin v0.12]
```
==> $PINS output :
```
[PIN:19,Feed hold]
[PIN:20,Cycle start]
[PIN:18,Emergency stop]
[PIN:28,Probe]
[PIN:9,X limit min]
[PIN:10,Y limit min]
[PIN:11,Z limit min]
[PIN:22,Aux in 0,P0]
[PIN:21,Aux in 1,P1]
[PIN:23,Aux in 2,P2]
[PIN:PIO2,X step]
[PIN:PIO3,Y step]
[PIN:PIO4,Z step]
[PIN:5,X dir]
[PIN:6,Y dir]
[PIN:7,Z dir]
[PIN:8,Steppers enable]
[PIN:12,Aux out 0,P0]
[PIN:26,Aux out 1,P1]
[PIN:15,Aux out 2,P2]
[PIN:14,Aux out 3,P3]
[PIN:27,LED adressable]
[PIN:1,RX,Primary UART]
[PIN:0,TX,Primary UART]
```
Line for spindle on/off command is missing, expecting : `[PIN:13,Spindle on]`
But spindle seems to be correctly defined :
==> $SPINDLES output :
```
0 - Basic
```
==> $SPINDLESH output :
```
[SPINDLE:0|0|1|*|Basic|0.0,1000.0]
```
And, of course, M3/M5 commands have no action on GPIO 13.

terjeio, 2025-01-01:
If you have a custom map file you have to modify it for new style assignments. See one of the other maps for how to.

terjeio, 2025-01-01:
Oops - I see now that the generic maps has not been updated...

setaperlacloche, 2025-01-01:
Thanks, I will wait for the generic maps update.

setaperlacloche, 2025-01-02:
Quickly tested. Perfect !
Many thanks to you.

closed 2025-01-02 by setaperlacloche