grblHAL/RP2040 #109: W5500 build with RP2350 failed

Dexus, 2025-01-20:
```
Command failed: make
In file included from /home/dev/RP2040/networking/networking.h:55,
from /home/dev/RP2040/networking/wiznet/enet.c:50:
/home/dev/RP2040/lwipopts.h:46: warning: "LWIP_CHKSUM_ALGORITHM" redefined
46 | #define LWIP_CHKSUM_ALGORITHM 3
|
In file included from /home/sdk/pico-sdk/lib/lwip/src/include/lwip/arch.h:48,
from /home/sdk/pico-sdk/lib/lwip/src/include/lwip/debug.h:40,
from /home/sdk/pico-sdk/lib/lwip/src/include/lwip/opt.h:52,
from /home/sdk/pico-sdk/lib/lwip/src/include/lwip/netif.h:40,
from /home/dev/RP2040/networking/wiznet/enet.c:31:
/home/dev/RP2040/networking/wiznet/arch/cc.h:96: note: this is the location of the previous definition
96 | #define LWIP_CHKSUM_ALGORITHM 0
|
/home/dev/RP2040/networking/wiznet/enet.c:176:38: warning: 'struct mdns_service' declared inside parameter list will not be visible outside of this definition or declaration
176 | static void mdns_device_info (struct mdns_service *service, void *txt_userdata)
| ^~~~~~~~~~~~
/home/dev/RP2040/networking/wiznet/enet.c: In function 'mdns_device_info':
/home/dev/RP2040/networking/wiznet/enet.c:181:5: warning: implicit declaration of function 'mdns_resp_add_service_txtitem' [-Wimplicit-function-declaration]
181 | mdns_resp_add_service_txtitem(service, "model=grblHAL", 13);
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dev/RP2040/networking/wiznet/enet.c: At top level:
/home/dev/RP2040/networking/wiznet/enet.c:186:39: warning: 'struct mdns_service' declared inside parameter list will not be visible outside of this definition or declaration
186 | static void mdns_service_info (struct mdns_service *service, void *txt_userdata)
| ^~~~~~~~~~~~
/home/dev/RP2040/networking/wiznet/enet.c: In function 'netif_status_callback':
/home/dev/RP2040/networking/wiznet/enet.c:238:9: warning: implicit declaration of function 'mdns_resp_init' [-Wimplicit-function-declaration]
238 | mdns_resp_init();
| ^~~~~~~~~~~~~~
/home/dev/RP2040/networking/wiznet/enet.c:240:29: warning: implicit declaration of function 'mdns_resp_add_netif' [-Wimplicit-function-declaration]
240 | if((services.mdns = mdns_resp_add_netif(netif_default, network.hostname, MDNS_TTL) == ERR_OK)) {
| ^~~~~~~~~~~~~~~~~~~
/home/dev/RP2040/networking/wiznet/enet.c:242:13: warning: implicit declaration of function 'mdns_resp_add_service' [-Wimplicit-function-declaration]
242 | mdns_resp_add_service(netif_default, network.hostname, "_device-info", DNSSD_PROTO_TCP, 0, MDNS_TTL, mdns_device_info, "version=" GRBL_VERSION);
| ^~~~~~~~~~~~~~~~~~~~~
/home/dev/RP2040/networking/wiznet/enet.c:242:84: error: 'DNSSD_PROTO_TCP' undeclared (first use in this function); did you mean 'IP_PROTO_TCP'?
242 | mdns_resp_add_service(netif_default, network.hostname, "_device-info", DNSSD_PROTO_TCP, 0, MDNS_TTL, mdns_device_info, "version=" GRBL_VERSION);
| ^~~~~~~~~~~~~~~
| IP_PROTO_TCP
/home/dev/RP2040/networking/wiznet/enet.c:242:84: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [CMakeFiles/grblHAL.dir/build.make:482: CMakeFiles/grblHAL.dir/networking/wiznet/enet.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1873: CMakeFiles/grblHAL.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

terjeio, 2025-01-20:
Its due to a typo in the Web Builder backend plus an incompatible mdns lwip app in the SDK. I'll have to add a check for the lwip version used and change the grblHAL code accordingly.
Anyway the Web Builder at least compiles the code now, I have yet to test if it still works.

terjeio, 2025-01-22:
Updated code [committed](https://github.com/grblHAL/Plugin_networking/commit/881d857e2dcea05afdb23d34f996638f2ca5dcc4).

closed 2025-01-22 by terjeio

Dexus, 2025-01-22:
@terjeio 

```
Command failed: make
In file included from /home/dev/RP2040/driver.c:115:
/home/dev/RP2040/wifi.h:28:10: fatal error: lwip/ip_addr.h: No such file or directory
28 | #include "lwip/ip_addr.h"
| ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/grblHAL.dir/build.make:90: CMakeFiles/grblHAL.dir/driver.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1954: CMakeFiles/grblHAL.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

terjeio, 2025-01-23:
Another backend issue fixed - it was not set up for the new Pico2 W. Was ok for Pico W.