grblHAL/RP2040 #111: Mapping causes grblHAL to appear not to run at all

true-greg-anderson2, 2025-01-24:
[Failure.zip](https://github.com/user-attachments/files/18539341/Failure.zip)

If I build the default, everything seems to work (with the generic map).
If I define everything inside the #ifdef MAKE_IT_FAIL inside the attached my_machine.h, including using 6 axis and the mappings in the attached mapping file, it does not appear to run.
There is no response on the usb/serial port when built with the attached files.

true-greg-anderson2, 2025-01-24:
Additional Notes:
I am running on a raspberrypi pico 2, using SDK 2.1.0

terjeio, 2025-01-24:
These #define statements cannot be set in _my_machine.h_ since they are for the core and are not visible for it:
```
#define DEFAULT_HOMING_ALLOW_MANUAL On
#define DEFAULT_HOMING_SINGLE_AXIS_COMMANDS On
#define DEFAULT_HOMING_USE_LIMIT_SWITCHES On
#define N_AXIS 6
#define DEFAULT_HOMING_CYCLE_0 (Y_AXIS_BIT)
#define DEFAULT_HOMING_CYCLE_1 (Z_AXIS_BIT)
#define DEFAULT_HOMING_CYCLE_2 (X_AXIS_BIT)
#define DEFAULT_HOMING_CYCLE_3 (A_AXIS_BIT | B_AXIS_BIT | C_AXIS_BIT)
```
Either modify [grbl/config.h](https://github.com/grblHAL/core/blob/4dc17753bb6bb7d5878ebe8d01bcea82e23e23b5/config.h) or add them to [CMakeLists.txt](https://github.com/grblHAL/RP2040/blob/master/CMakeLists.txt) with `target_compile_definitions()`.

Do not `#define M3_AVAILABLE - M5_AVAILABLE` in the map file, these will be defined elsewhere depending on the configuration in _grbl/config.h_ and _my_machine.h_. Follow the same structure as used in the other map files.

true-greg-anderson2, 2025-01-24:
Thank you - after some more reading I figured that out and corrected that problem - but it was not the cause.
The cause appears to be using the STEP_PORT GPIO_PIO_1.
I was able to trace that it stopped execution within the call assign_step_sm(&x_step_pio, &x_step_sm, X_STEP_PIN);  (or there abouts).
(line 2967 in driver.c).  When I changed to use GPIO_PIO it started working again.