grblHAL/RP2040 #112: Ethernet W5500 and SD Card stop responding after card inserted

robertctrescott, 2025-01-26:
This system is running the latest grblHAL and plug-ins firmware on the PicoCNC board and compiled in VSCode.
The SPI bus signals appear correct on the scope, while running ethernet, but after SDCard is inserted they begin to falter and stop altogether after a few seconds. 

The hardware looks like it's addressing each CS line correctly.
SD Card works properly using USB only, without ethernet module installed.
Ethernet works properly until SD card is inserted.

Unsure if this is relevant, but the SD Card SPI CS line goes active low for 5ns during Ethernet transactions, corresponding to some transitions of MOSI and MISO. This appears to be a 5ns glitch which may interfere with SD Card logic. Not sure??? However, I'm searching for more ways to isolate the actual problem.

terjeio, 2025-01-26:
Which board?

robertctrescott, 2025-01-26:
PicoCNC.

terjeio, 2025-01-26:
Ok, I have been running a 3.5MB job successfully via the WebUI a few times since I last commented - with a Pico2.
I have seemingly random failures (error 64) when unmounting/remounting the card from the WebUI, but I have yet to see any hang.
I'll try with a Pico to see if that behaves differently.

terjeio, 2025-01-26:
The Pico seems to run well, testing now with ioSender connected via Telnet and running a job on the SD card.
Which grblHAL build do you have (post the `$I` output)?
Is it a Web Builder build?
If the card is inserted when powering up the board and automount is enabled ($650=1 or 3) does it still fail?

robertctrescott, 2025-01-26:
If the card is already inserted when powering up, the network does not crash with $650=1. 
However, with $650 =1, if the card is inserted after powering up, ioSender crashes with an Unhandled exception: Unable to read data...
The board must also be restarted after this happens.

As I mentioned in my original post, this build is compiled in VSCode, it is not a Web Builder build.

$I printout follows below:

*** Connected to GRBL 1.1f
>>> $I
[VER:1.1f.20250120:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,ES,SED,RTC,ETH,MBIO,SD,YM]
[FIRMWARE:grblHAL]
[SIGNALS:HSEP]
[NVS STORAGE:*FLASH 4K]
[FREE MEMORY:121K]
[DRIVER:RP2040]
[DRIVER VERSION:250102]
[DRIVER OPTIONS:SDK_2.1.0]
[BOARD:PicoCNC]
[AUX IO:1,7,0,0]
[WIZCHIP:W5500]
[MAC:00:08:dc:12:34:56]
[IP:192.168.4.80]
[NETCON:Telnet]
[PLUGIN:MODBUS v0.18]
[PLUGIN:Robs Blinky LED v1.00]
[PLUGIN:MODBUS IO v0.1]
[PLUGIN:SDCARD v1.19]
[PLUGIN:FS macro plugin v0.15]
ok

terjeio, 2025-01-26:
I finally managed to persuade VSCode to use SDK 2.1.0 - not easy...
Still, with the updated SDK, I am not able to replicate: power up the controller with the sdcard out, connect ioSender via Telnet, insert the card and open the _SD Card_ tab and the files are there and I am able to run them.

This is with the latest commit and ioSender 2.0.45.
I see that you have some third party plugins added to the build - could they somehow interfere?
So please try a Web Builder build if possible. Still failing?

robertctrescott, 2025-01-26:
Thanks for confirming that everything should work together.
I disabled the third-party plug-ins and the ioSender 2.0.45 crashed from the ethernet becoming unresponsive.
I loaded up the Web Builder firmware.uf2 with the proper settings for my PicoCNC hardware. 
It looked like it was working for a few SD Card reads, but it failed as ioSender reported the connection was closed by the remote host.

I rebuilt my original system and found if I just click a few times on the checkbox to read all files, ioSender would crash... without an SD Card installed!  During this test, GDB was running and stopped on an isr_hardfault. The screenshot below describes the CALL STACK.

[image]

terjeio, 2025-01-28:
No matter what I try, I have not been able to replicate the behaviour you have.
Anyway I have changed the SD card code a bit that _could_ help, and I'll soon make an ioSender edge version available that should interact with the SD card in a better way. The changed controller code has now been committed.

When testing I have managed to crash ioSender on occasion leaving the controller still wired for Telned communication. The controller seems to be unresponsive when that happens, but I can connect via USB/putty and I get responses when I send real time report requests. Currently there is no way other than a hard reset to restore communication. Only on one occasion did I manage to crash the controller (hardfault) - this was while testing code changes, and I was not able to replicate the crash. So the controller code seems to be pretty robust, the main issue it beeing left in an unrepsonsive state when a Telnet client crashes without disconnecting. I am not sure if it is going to be possible to fix this.

BTW the stack trace you provided does not help much - other than the crash happened when trying to write to the output stream(s).

terjeio, 2025-01-28:
The new ioSender [edge version](https://github.com/terjeio/ioSender#edge-pre-releases) is now available.

robertctrescott, 2025-01-28:
I'm also using a 32GB SanDisk. It's an Ultra microSD HC I, C10. Also using a PNY 16GB Elite HC I.
They are formatted FAT32 and both read fine while in USB.

Interestingly, however, with $650 = 1 setting, I could get my system to crash ethernet, just by accessing the SD Card tab!
Without even inserting the sd card. 

To get some stability with ethernet, I disabled the SD Card in my_machine.h. That brought on a whole other can of worms.
Pico_CNC.c and Pico_CNC_map.h required some changes, to permit an ethernet build without SD Card enabled. 
I got it all sorted out with all the proper SPI CS and WZ_RST... I haven't seen any other issues or discussions about this Pico_CNC product, which gives me some concern... I can't be the only one who's run into this?

robertctrescott, 2025-01-29:
You got it!
SD Card works as expected while in ethernet with no crashes.
From my standpoint, I believe you have closed this issue.
Thank you for your perseverance!