grblHAL/RP2040 #116: How to configure the min and max switch

didecibel, 2025-02-28:
Hello, I can't figure out how to configure the min and max limit switches. And what are the pins. I am not fluent in C so... 
This is on a rpi pico if someone could help me. It seams that is done via the driver.c file but I don't know how.
Thanks a lot.

terjeio, 2025-03-01:
It is done in the [map file by assigning pins to the function](https://github.com/grblHAL/RP2040/blob/626cf574f5914d01a63aae03e3bcb9a10a71099a/boards/generic_map.h#L50-L53).
E.g. `#define X_LIMIT_PIN_MAX 12`

Easiest is to copy an existing map file to _my_machine_map.h_, modify it and enable that in [my_machine.h](https://github.com/grblHAL/RP2040/blob/626cf574f5914d01a63aae03e3bcb9a10a71099a/my_machine.h#L35).

terjeio, 2025-03-25:
@hanke-cnc If the home switch is separate from limit switches you may need three, home min, limit min and max. Grbl/grblHAL can combine home and limit inputs and use soft limit instead of limit max so only one is required (unless auto-squaring is enabled). Commercial controllers may _require_ three for practical and/or safety reasons?
FYI the grblHAL core supports separate home and limit switches, but not all drivers and no(?) board maps do.