grblHAL/RP2040 #119: grblHAL compilation

didecibel, 2025-03-05:
Hello,
the compilation doesn't work. I appears to be because of a newer version of picotool and pico-sdk that are 2.1.1 and the scripts call for 2.1.0. 
I have tried many combination but after a lot of installations and tweeking of the cmake files I am now lost.
There is nothing on the web that explains how to compile the grblHAL for rpi pico.
The few that exist look old and none of them works. 
This would be great to have a compile with terminal tutorial.

terjeio, 2025-03-05:
Which scripts? Which development environment?
I am developing with VSCode using 2.1.1.

terjeio, 2025-03-05:
So ninja is missing from the installation?
I run VSCode on a Win10 machine and ninja.exe exists in a path similar to what the error message above contains.
The Web Builder runs on a ubuntu server and uses a build environment set up using [this description](https://github.com/JacobBeningo/RpiPico), and that does not contain a ninja directory.
I do not have a Linux desktop machine so I have no idea what the .pico-sdk directory should contain.

BTW the repo was recently updated to use SDK 2.1.1 - perhaps you have to pull the [latest version](https://github.com/grblHAL/RP2040/blob/a9ca78a8a98e6e89aa292b1a763b285de153ad4a/CMakeLists.txt#L1-L13).

didecibel, 2025-03-05:
Ok, I managed to do it. Sorry for this useless issue. I was not using vscode well.

closed 2025-03-05 by didecibel

mikel1999, 2025-03-06:
> Ok, I managed to do it. Sorry for this useless issue. I was not using vscode well.

Can you tell me the solution? I'm facing the same problem.

didecibel, 2025-03-08:
You have to install the pico library in vscode, open the RP2040 folder and then build the grblhal.uf2 by clicking on the build button on the downleft side of the interface.