grblHAL/RP2040 #121: Compile error Visual Studio Code

mikel1999, 2025-03-06:
Hello! I am a beginner in compiling with raspberry pico. I am getting this error, even though I have installed the Ninja package. I think it is helpful for many users.

[image]

I mention that I used the specified versions: SDK 2.1.0 / toolchainVersion 13_2_Rel1.Any help?

mikel1999, 2025-03-14:
Nope. I try to install VSCode and plugin(for pico) . I try with another pc and i can't find the solution . Every time I try to compile , the result it' same . I try to install extra Ninja plugin , but without succes. Maybe someone can help me. If I manage to solve it I could make a tutorial on modifying the files for beginners.

terjeio, 2025-03-14:
I just installed VSCode and the Pico plugin on a Win10 Pro machine with a newly installed OS (and boy how MS has crapified the UI experience in the latest version - that annoying paperclip, Clippy?, is back times ten).

I had to restart VSCode after installing the plugin due to it complaning about python missing and some other messages I did not catch, after restarting the plugin was able to complete its installation. I then opened the folder with the RP2040 driver and was able to compile.

I am not able to provide support for VSCode or Pico plugin installations since I do not have the faintest idea about what is going on behind the scenes. Issues with the plugin should be reported [here](https://github.com/raspberrypi/pico-vscode).

mikel1999, 2025-03-20:
Thanks for the help. I solved the problem.

closed 2025-03-20 by mikel1999

hanke-cnc, 2025-03-24:
> Thanks for the help. I solved the problem.

Hello, how did you resolve this issue? I encountered the same error. I can successfully create and compile a blinking LED project, but I can't compile this one.