grblHAL/RP2040 #122: Virtual UART and TXEN for RS485

2lostkiwis, 2025-03-20:
Steps to reproduce:

Compile firmware from [https://svn.io-engineering.com/](https://svn.io-engineering.com/) updated to grblHAL build/patch 20250312
Change connection to (Virtual) UART

I am debugging this with just a bare PICO, not connected to a shield/board.

I can verify that I can communicate over the UART interface by connecting an FTDI board to PICO pins GPIO0 (TX) and GPIO1 (RX).
e.g. ? returns the following from the UART:
`<Alarm|MPos:0.000,0.000,0.000|Pn:XYZHSEP|A:|TLR:0>`

Now I would like to connect an RS485 transceiver to this port, this uses the same pins along with GPIO2 (TXEN) to enable the RS485 transmitter.
The problem I have is that, this line forever sits low, when the PICO is transmitting a serial response, it does not enable the RS485 transmitter.

I am not sure if this is a bug in the code, or perhaps there are some deeper configuration options not exposed by the web builder interface.
Am I correct that I should be able to connect an RS485 transceiver to the UART interface?

terjeio, 2025-03-20:
> I am not sure if this is a bug in the code, or perhaps there are some deeper configuration options not exposed by the web builder interface.

Option is here for RS-485 breakouts that do not support auto direction switching:

[image]

terjeio, 2025-03-21:
> I want to use the TXEN for talking to the UART when used for sending gcode commands.

You have implemented a half-duplex based protocol for communicating with the controller?

2lostkiwis, 2025-03-21:
> You have implemented a half-duplex based protocol for communicating with the controller?

Yes, absolutely :-)

terjeio, 2025-03-21:
OK, then you have to add control of the pin to your protocol code. And you cannot use the Web Builder to build custom code unless it is in board specific code for a published board map. And just to be clear - the standard Grbl/grblHAL protocol is _not_ half-duplex. To transmit that over RS-485 you need two transceivers at each end (four wires).

2lostkiwis, 2025-03-23:
Thanks Terjeio, we can mark this as closed then. Many thanks for your help.

closed 2025-03-23 by 2lostkiwis