grblHAL/RP2040 #124: A bit strange?

hanke-cnc, 2025-03-23:
I am considering replacing the next-generation chip for my control board. I referred to the prices of stm32H7 and RP2350 chips. RP2350 can be bought for $1 in China. The chip parameters also look good, and its various specifications seem better than f407. However, I tested RP2350, and it can only stably maintain an output of 150kHz at most. I feel that I cannot reverse upgrade, and I cannot explain this to the users.

I just want to know why this is. Did the RP2350 use inferior peripherals, such as timers? Is that why it's so cheap?😄

terjeio, 2025-03-24:
> Did the RP2350 use inferior peripherals, such as timers?

No, it is superior for CNC use due to the PIOs - makes it easy to generate high precision step pulses with relatively little jitter (only one interrupt per pulse compared to two or three for all the other drivers except the ESP32 when using RMT).
Which build did you test with and with which stepper settings?

hanke-cnc, 2025-03-24:
> No, it is superior for CNC use due to the PIOs - makes it easy to generate high precision step pulses with relatively little jitter (only one interrupt per pulse compared to two or three for all the [...]

Oh, sorry, yesterday I tried for a long time but couldn't successfully compile this project using vscode, so I closed this issue.

Since I did not successfully compile this project, I used the firmware obtained from your web-based compiler. Using an oscilloscope to test, it can only stably maintain 150kHz and is somewhat unstable at 200kHz.

terjeio, 2025-03-24:
Which board map?

hanke-cnc, 2025-03-24:
> Which board map?

I forgot, it seems to be <PicoCNC w/RP2350>. Of course, you can also give me a test firmware for the rp2350. I am very interested in the $1 rp2350 chip; it is very suitable for replacing and upgrading the stm32f407.

terjeio, 2025-03-24:
> I forgot, it seems to be <PicoCNC w/RP2350>

If so this uses a shift register for step output, no the fastest option.

>  Of course, you can also give me a test firmware for the rp2350.

Try with the RP23U5XBB board from the Web Builder.
FYI I have added limiting of minimum step off length to 2 us, so 250Khz will be the max default step rate. I currently working on adding this to the STM32 drivers as well. I will add overridable compile time settings to allow tuning of step pulse length and off time < 2us to allow use with high performance stepper drivers.
And another FYI is that I am about to check how much moving some IRQ handlers to RAM will affect max step rates for the STM32 drivers (except the F103 which has little RAM to play with).

closed 2025-03-24 by hanke-cnc