grblHAL/RP2040 #125: It seems that M5 cannot turn off M4's pins?

hanke-cnc, 2025-03-25:
All the code hasn't been changed; only the pins m3 and m4 were modified for testing. I tested it twice, and it seems to be the same each time.

Furthermore, I don't quite understand your handling of "M3 and M4." According to the LinuxCNC documentation, M3 and M4 should be mutually exclusive. That is, sending "M3 S100" should deactivate M4, and sending "M4" should deactivate M3. You are using M3 as the spindle enable signal and M4 as the spindle direction signal, but based on the LinuxCNC documentation, it doesn't seem like it should be the case. Of course, I haven't used LinuxCNC.  https://linuxcnc.org/docs/stable/html/gcode/m-code.html#mcode:m3-m4-m5

[RP2350B_5X_map.h.txt](https://github.com/user-attachments/files/19443684/RP2350B_5X_map.h.txt)

hanke-cnc, 2025-03-25:
That's strange. I just tested it, and it's working again. It's puzzling.

terjeio, 2025-03-25:
> You are using M3 as the spindle enable signal and M4 as the spindle direction signal

M3 and M4 turns on spindle enable, M5 turns it off.
M3 and M4 controls spindle direction, M5 not.

Perhaps confused about M5 not resetting the direction signal?

hanke-cnc, 2025-03-25:
> > You are using M3 as the spindle enable signal and M4 as the spindle direction signal
> 
> M3 and M4 turns on spindle enable, M5 turns it off. M3 and M4 controls spindle direction, M5 not.
> 
> Perhaps confused about M5 not resetting the direction signal?

Yes, it seems that because M5 doesn't turn off M4, it caused my brain to almost fall into confusion again.

closed 2025-03-25 by hanke-cnc