grblHAL/RP2040 #129: strange bug encountered. dir-pins

keverin16, 2025-04-09:
hello everyone, i encountered a strange bug.
I compiled rp2040 firmware invisual studio, only changed to generic 4-axis, changed some pin numbers and set the number of axes to 4 in grbl-config. after compiling it marks "M3_STEP_PIN STEP_PINS_BASE + 3" yellow in visual studio, but i couldnt find a problem. 
Anyway, after connecting to the board with openbuilds control i get ERROR 71, but it does connect ok. Does not want to connect to gsender, shows incompatible.
Now everything works fine in openbuilds control, except direction signal. I can jog+, i can jog-, it doesnt change. Only way i can make it change is if i invert the axis with $3. So it seems to be configured right. Anyone has an idea?

terjeio, 2025-04-09:
Post your modified board map please, or replicate the bug with one of the provided board maps and tell me which.

keverin16, 2025-04-10:
> Post your modified board map please, or replicate the bug with one of the provided board maps and tell me which.

i tried with the stock config - compiles without problems
then i changed to board_generic_4axis by uncommenting this in my_machine - compiles without problems
then i changed the config to real 4-axes in grbl - config - N_Axis to "4" - then i get warnings in the driver.h folder. it underlines this in yellow: "M3_STEP_PIN STEP_PINS_BASE + 3"

terjeio, 2025-04-10:
> it underlines this in yellow: "M3_STEP_PIN STEP_PINS_BASE + 3"

Oops, a duplicate definition - will fix in the next commit. You can ignore it for now, or delete one of them.

keverin16, 2025-04-11:
> > it underlines this in yellow: "M3_STEP_PIN STEP_PINS_BASE + 3"
> 
> Oops, a duplicate definition - will fix in the next commit. You can ignore it for now, or delete one of them.

I removed it, now that problem is gone, thanks.
But the main problem is still here and i have no idea why its here
all direction pins never change their state when i change direction. But when i invert them with $3 i can change them one by one, so they are working. Also still get error 71 when i connect with openbuilds control.

keverin16, 2025-04-11:
here is what i see when connecting to openbuilds control:

[10:45:40] [ connect ] PORT INFO: Port is now open: COM3 - Attempting to detect Controller...

[10:45:40] [ connect ] Attempting to detect Controller (1): (Autoreset)

[10:45:40] [ ] ok

[10:45:40] [ connect ] Attempting to detect Controller (2): (Ctrl+X)

[10:45:41] [ connect ] Attempting to detect Controller (3): (others)

[10:45:41] [ ] error:71

[10:45:41] [ connect ] Attempting to detect Controller (4): (DTR Enable)

[10:45:41] [ ] GrblHAL 1.1f ['$' or '$HELP' for help]

[10:45:41] [ ] GrblHAL 1.1f ['$' or '$HELP' for help]

[10:45:42] [ connect ] Firmware Detected: grblHAL version 1.1f dated on COM3

[10:45:42] [ [external from hardware] ] OpenBuilds CONTROL received a FEEDHOLD notification from Grbl: This could be due to someone pressing the HOLD button (if connected)

[10:45:42] [ [external from hardware] ] OpenBuilds CONTROL received a CYCLESTART/RESUME notification from Grbl: This could be due to someone pressing the CYCLESTART/RESUME button (if connected)

[10:45:42] [ $$ ] $0=5.0 ;Step pulse time, microseconds

[10:45:42] [ $$ ] $1=25 ;Step idle delay, milliseconds

[10:45:42] [ $$ ] $2=0 ;Step pulse invert, mask

[10:45:42] [ $$ ] $3=0 ;Step direction invert, mask

[10:45:42] [ $$ ] $4=15 ;Invert step enable pin, boolean

[10:45:42] [ $$ ] $5=0 ;Invert limit pins, boolean/mask

[10:45:42] [ $$ ] $6=0 ;Invert probe pin, boolean

[10:45:42] [ $$ ] $9=1 ;PWM Spindle as bitfield where setting bit 0 enables the rest

[10:45:42] [ $$ ] $10=511 ;Status report options, mask

[10:45:42] [ $$ ] $11=0.010 ;Junction deviation, millimeters

[10:45:42] [ $$ ] $12=0.002 ;Arc tolerance, millimeters

[10:45:42] [ $$ ] $13=0 ;Report in inches, boolean (CONTROL needs $13=0)

[10:45:42] [ $$ ] $14=0 ;Invert control input signals, mask

[10:45:42] [ $$ ] $15=0 ;Coolant pins invert, mask

[10:45:42] [ $$ ] $16=0 ;Spindle pins invert, mask

[10:45:42] [ $$ ] $17=0 ;Control pins pullup disable, mask

[10:45:42] [ $$ ] $18=0 ;Limit pins pullup disable, mask

[10:45:42] [ $$ ] $19=0 ;Probe pin pullup disable, boolean

[10:45:42] [ $$ ] $20=0 ;Soft limits enable, boolean

[10:45:42] [ $$ ] $21=0 ;Hard limits enable, boolean

[10:45:42] [ $$ ] $22=0 ;Homing cycle enable, boolean (Grbl) / mask (GrblHAL)

[10:45:42] [ $$ ] $23=0 ;Homing direction invert, mask

[... 128 lines cut ...]
[10:45:43] [ $I ] [NEWOPT:ENUMS,RT+,ES,EXPR,SED,RTC]

[10:45:43] [ $I ] [FIRMWARE:grblHAL]

[10:45:43] [ $I ] [SIGNALS:HSEP]

[10:45:43] [ $I ] [NVS STORAGE:*FLASH 4K]

[10:45:43] [ $I ] [FREE MEMORY:216K]

[10:45:43] [ $I ] [DRIVER:RP2040]

[10:45:43] [ $I ] [DRIVER VERSION:250405]

[10:45:43] [ $I ] [DRIVER OPTIONS:SDK_2.1.1]

[10:45:43] [ $I ] [AUX IO:2,0,0,0]

[10:45:43] [ $I ] [PLUGIN:Bootloader Entry v0.01]

[10:45:43] [ $I ] ok

[10:45:43] [ $G ] [GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]

[10:45:43] [ $G ] ok

keverin16, 2025-04-11:
i am really at a loss here, i measure directly at the rp2040 GPIO pin. so with $3 i can switch every direction pin between 0V and 3.3V. But if i jog around in both directions they just never change their state.

terjeio, 2025-04-11:
I have just tested a 4-axis config made by the Web Builder and that works, so I need [this](https://github.com/grblHAL/RP2040/issues/129#issuecomment-2790477012) or I am not able provide any help. And BTW which version do you have - $I output please.

Error 71 is from a bad expression - due to openbuilds control sending an incorrect command?

keverin16, 2025-04-11:
> I have just tested a 4-axis config made by the Web Builder and that works, so I need [this](https://github.com/grblHAL/RP2040/issues/129#issuecomment-2790477012) or I am not able provide any help. [...]

/*
  generic_map_4axis.h - driver code for RP2040 ARM processors

  Part of grblHAL

  Copyright (c) 2021-2025 Terje Io
  Copyright (c) 2021 Volksolive

  grblHAL is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  grblHAL is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  along with grblHAL. If not, see <http://www.gnu.org/licenses/>.
*/

#if TRINAMIC_ENABLE
#error Trinamic plugin not supported!
#endif

#if N_ABC_MOTORS > 1
#error "Axis configuration is not supported!"
#endif

// Define step pulse output pins.
#define STEP_PORT               GPIO_PIO  // N_AXIS pin PIO SM
#define STEP_PINS_BASE          11         // N_AXIS number of consecutive pins are used by PIO

// Define step direction output pins.
#define DIRECTION_PORT          GPIO_OUTPUT
#define X_DIRECTION_PIN         15
#define Y_DIRECTION_PIN         16
#define Z_DIRECTION_PIN         17
#define DIRECTION_OUTMODE       GPIO_SHIFT6

// Define stepper driver enable/disable output pin.
#define ENABLE_PORT             GPIO_OUTPUT
#define STEPPERS_ENABLE_PIN     7

// Define homing/hard limit switch input pins.
#define X_LIMIT_PIN             1
#define Y_LIMIT_PIN             2
#define Z_LIMIT_PIN             3
#define LIMIT_INMODE            GPIO_MAP

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_DIRECTION_PIN        (Z_DIRECTION_PIN + 1)
#define M3_LIMIT_PIN            (Z_LIMIT_PIN + 1)
#else
#define AUXINPUT0_PIN           14
#define AUXOUTPUT0_PORT         GPIO_OUTPUT
#define AUXOUTPUT0_PIN          5
#define AUXOUTPUT1_PORT         GPIO_OUTPUT
#define AUXOUTPUT1_PIN          9
#endif

#define AUXOUTPUT2_PORT         GPIO_OUTPUT // Spindle PWM
#define AUXOUTPUT2_PIN          22
#define AUXOUTPUT3_PORT         GPIO_OUTPUT // Spindle direction
#define AUXOUTPUT3_PIN          9
#define AUXOUTPUT4_PORT         GPIO_OUTPUT // Spindle enable
#define AUXOUTPUT4_PIN          19   
#define AUXOUTPUT5_PORT         GPIO_OUTPUT // Coolant flood
#define AUXOUTPUT5_PIN          20   
#define AUXOUTPUT6_PORT         GPIO_OUTPUT // Coolant mist
#define AUXOUTPUT6_PIN          21   

// Define driver spindle pins
#if DRIVER_SPINDLE_ENABLE
#define SPINDLE_PORT            GPIO_OUTPUT
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_PWM
#define SPINDLE_PWM_PIN         AUXOUTPUT2_PIN
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_DIR
#define SPINDLE_DIRECTION_PIN   AUXOUTPUT3_PIN
#endif
#if DRIVER_SPINDLE_ENABLE & SPINDLE_ENA   
#define SPINDLE_ENABLE_PIN      AUXOUTPUT4_PIN
#endif

// Define flood and mist coolant enable output pins.
#if COOLANT_ENABLE
#define COOLANT_PORT            GPIO_OUTPUT
#endif
#if COOLANT_ENABLE & COOLANT_FLOOD
#define COOLANT_FLOOD_PIN       AUXOUTPUT5_PIN
#endif
#if COOLANT_ENABLE & COOLANT_MIST
#define COOLANT_MIST_PIN        AUXOUTPUT6_PIN
#endif

#define AUXINPUT1_PIN           23
#define AUXINPUT2_PIN           24
#define AUXINPUT3_PIN           5 // Probe

// Define user-control controls (cycle start, reset, feed hold) input pins.
#define RESET_PIN               8
#define FEED_HOLD_PIN           6
#define CYCLE_START_PIN         25

#if PROBE_ENABLE
#define PROBE_PIN               AUXINPUT3_PIN
#endif

#if SAFETY_DOOR_ENABLE
#define SAFETY_DOOR_PIN         AUXINPUT1_PIN
#elif MOTOR_FAULT_ENABLE
#define MOTOR_FAULT_PIN         AUXINPUT1_PIN
#endif

keverin16, 2025-04-11:
> which version do you have - $I output please.

what do you mean by that? Which version of what?

terjeio, 2025-04-11:
> #define DIRECTION_OUTMODE GPIO_SHIFT6

cannot work - try `GPIO_SHIFT15` instead, since you have changed the pins

keverin16, 2025-04-11:
Great, will try tomorrow. 
I am far from a programmer, can you explain what that number 15 means ond how do i know which to pick when i change the pin numbers?

terjeio, 2025-04-11:
It is number of [bit shifts](https://stackoverflow.com/questions/141525/what-are-bitwise-shift-bit-shift-operators-and-how-do-they-work) to get the core supplied direction bits aligned with the GPIO register pins. Shift equals the X-dir pin, and shifting can only be used when the pins are consecutive (X, Y, Z, ... i.e. in the same order as delivered from the core).