grblHAL/RP2040 #133: For input signals such as Start, Pause, Reset, and Emergency Stop (E-Stop), if their assigned GPIO pin number (or digital definition) exceeds a certain value, triggering these signals will cause a crash.

hanke-cnc, 2025-05-20:
"After pulling the latest source code and compiling it, I discovered an issue on my custom-made RP2354 control board. For signals such as start, pause, and emergency stop, if the GPIO pin assigned to them is defined with a number (or GPIO index) greater than a certain value, triggering the signal causes the program to crash and become completely unresponsive, offering no feedback whatsoever."

"To confirm if this was the cause, or to further pinpoint the problem, I swapped the original X-axis limit GPIO_23 with the original ESTOP pin GPIO_38 in an experiment. There were no issues."

[image]

hanke-cnc, 2025-05-20:
"I'm guessing it's because you don't have an RP2354 development board?"

hanke-cnc, 2025-05-20:
"I've already experimented with the output side. The M3/M4/M7/M8 signals, defined on GPIO44-47, have no issues. The problem seems to be with signals like Start, Pause, and Emergency Stop, doesn't it?"

terjeio, 2025-05-20:
> greater than a certain value,

Which? I have just tested with pin 41 and 46 and those works for me.
And which "board" are you compiling for? I am using "pimoroni_pga2350".

> The M3/M4/M7/M8 signals, defined on GPIO44-47, have no issues. 

These are inputs, far easier to handle than inputs requiring an interrupt handler.

hanke-cnc, 2025-05-21:
> Which? I have just tested with pin 41 and 46 and those works for me. And which "board" are you compiling for? I am using "pimoroni_pga2350".
>
>
> These are inputs, far easier to handle than inputs [...]

"Oh, I re-tested it today. If I select the same "pimoroni_pga2350" as you, it works normally. Initially, I had selected "waveshare_rp2350_plus_4mb"."  

"Additionally, my board is populated with a W25Q32JVSSIQ 32Mbit flash chip. If I choose the "pimoroni_pga2350" configuration (the same as yours), I'm unclear if any modifications are necessary."