grblHAL/RP2040 #134: The SD card functionality seems to be particularly picky about the pins?

hanke-cnc, 2025-05-22:
Initially, in my control board design for the SD card, I used CS=5, CMD=4, SCK=3, DATA0=2, with a 10K resistor pulling up the CS pin, but it wouldn't work no matter what I tried.
#define SPI_PORT                 1
 #define SPI_SCK_PIN             3
 #define SPI_MOSI_PIN            4
 #define SPI_MISO_PIN            2

 #if SDCARD_ENABLE
 #define SD_CS_PIN               5
 #endif

Then, I had to use another blank PCB to solder the SD card and wire it to an RP2350 development board for testing. I achieved success with the configuration below and was able to successfully read the contents of the SD card. Strangely enough, the 10K resistor needed to be used as a pull-up on the MISO line, not on the CS pin.

// #define SPI_PORT                1
// #define SPI_SCK_PIN             26
// #define SPI_MOSI_PIN            27
// #define SPI_MISO_PIN            28

// #if SDCARD_ENABLE
// #define SD_CS_PIN               29
// #endif

OK, after succeeding with pins 26-29, I went back to test pins 2-5. Whether I tried swapping the various pins or attempting to use 10K pull-up resistors on each of these pins individually, I still couldn't get it to work. I'm wondering if I need to simultaneously use 10K pull-up resistors on multiple pins.

hanke-cnc, 2025-05-22:
I choose to use the pimorini_pga2350 board for compilation.

hanke-cnc, 2025-05-22:
I have purchased an SD card module today, which has a 10K pull-up resistor for each pin. The delivery will arrive in about 2 days. I strongly suspect that my lack of success is because each pin needs a 10K pull-up resistor.

[image]

terjeio, 2025-05-23:
I have not designed any boards for the Pico, however I have tested the driver with both the PicoCNC and RP23U5XBB board and those works fine - even when sharing the SPI bus with a WizChip ethernet module. And yes, you need pullup resistors on more than one line.

hanke-cnc, 2025-05-26:
> I have not designed any boards for the Pico, however I have tested the driver with both the PicoCNC and RP23U5XBB board and those works fine - even when sharing the SPI bus with a WizChip ethernet module. And yes, you need pullup resistors on more than one line.

Hi, just reporting:
I haven't had success using pins 2-5, even with an SD card module that has 10K pull-ups on all signals.
I measured the SD_SCK pin and saw a familiar voltage of around 2.1V. I've heard about the RP2350-E9 issue.
I tried pulling this pin down with a 4.7K resistor, but still didn't succeed, even though the 2.1V voltage is now gone from that pin.
I may need to make PCB modifications, or revert to the reliable F4 chip.

hanke-cnc, 2025-05-30:
> I have not designed any boards for the Pico, however I have tested the driver with both the PicoCNC and RP23U5XBB board and those works fine - even when sharing the SPI bus with a WizChip ethernet module. And yes, you need pullup resistors on more than one line.

Hello, I'm so sorry. There's absolutely nothing wrong with your code. I should take the blame for my own foolishness; I mistakenly thought SPI, like I2C, could use software emulation and therefore use any pins. Today, I checked the datasheet, adjusted the pin definitions, and the SD card is now working smoothly on pins 2-5.