grblHAL/RP2040 #135: Firmware compilation for RP2350(Pico2)

Deltaplc, 2025-05-23:
Good morning
Sorry to bother you...
I designed a board to host the RP2040... I successfully compiled the firmware with Visual Studio Code and installed it!
I used "my_machine_map.h" as pin map and I removed only the input probe, reset, feed and start pins... because I don't need them. I'm doing the first tests,
It seems that everything works correctly!
Question:
I would like to try to mount the new RP2350, can I install the same build file "grblHAL.uf2" that I installed on the RP2040 or do I have to change something during the compilation phase with Visual Studio?
Sorry for the question, but since I'm not a programming expert and it's my first time using a Raspberry, for me it's a bit out of my comfort zone.....

Thank you in advance for your support.
Best regards

terjeio, 2025-05-23:
You cannot flash the RP2040 version on a RP2350. Click on _Board:pico_ in the lower right corner to open a menu for selecting the processor, select pico2 (or pico2_w) and answer no to _Use Risc-V_,  and then recompile:

[image]

Deltaplc, 2025-05-23:
> You cannot flash the RP2040 version on a RP2350. Click on _Board:pico_ in the lower right corner to open a menu for selecting the processor, select pico2 (or pico2_w) and answer no to _Use Risc-V_, and then recompile:

OK, so the firmware is already ready to work with RP2350, even using "my_machine_map.h" ?
No need to change anything ?
Just select the type "board: Pico2" on the Visual Studio compiler....?
Did I understand correctly?
As always, thank you so much for your support!!

terjeio, 2025-05-23:
> OK, so the firmware is already ready to work with RP2350, even using "my_machine_map.h" ?

Correct, just compile for Pico 2.

Deltaplc, 2025-05-23:
> Correct, just compile for Pico 2.

Perfect!!
Thank you so much!!
Regards